FD-/RPiPlay #223: Black Screen RaspberryPi 3B+

Krathyn, 2021-02-18:
I can connect in the sense that it sees it based on the name but I am unable to get it to actually display anything it shows connected and I get a black screen. Is there an option I am missing or maybe I’ve compiled it wrong?

joaodriessen, 2021-03-24:
I’m having the same issue.
Run with the option “-b off” and the screen won’t go black.

unfortunately I get no video or audio.
I will update this post later with more detailed info on my system (and hopefully a log of rpiplay)

rogercorrea, 2021-05-24:
Same issue here.
My airplay when run using "-b auto" I can select in my iPhone but my TV goes black. Running using "-b off" my TV doesn't black but I can't select my device in iPhone airplay.

rogercorrea, 2021-05-25:
I tried use Youtube or another app to video play and my TV goes black. Using "mirror display" in my iPhone, I can work fine.