FD-/RPiPlay #224: A very useful change (for Zoom  users ) to video_renderer_gstreamer.c   (posted to UxPlay variant of RPiPlay)

fduncanh, 2021-02-21:
https://github.com/antimof/UxPlay/pull/31

This sets the gstreamer  video X-window title so Zoom recognizes the window, and allows screen sharing in a Zoom
session.   (This change is very useful for online teaching with Zoom, on desktop linux, using an ipad)

It is easily backported to RPiPlay.  (I've done it on my copy)
(UxPlay is an apparently unmaintained fork of RPiPlay, used by some linux-based teachers,  and its essential
content (video_renderer_gstreamer.c)  is already backported here.

daljit97, 2021-03-05:
@fduncanh what exactly needs to be done to support this?

fduncanh, 2021-03-25:
here is a patch against the current rpiplay for this.  its david venturas work on UxPlay, brought bto RPiPlay

[RPiPlay.patch.txt](https://github.com/FD-/RPiPlay/files/6202421/RPiPlay.patch.txt)

MichaelRenMR, 2022-01-25:
The patch didn't work for me on the most recent version of the code. An alternative solution (for XWin) I found was to use this command, which doesn't require modifying the source: 
```$ xdotool selectwindow set_window --name "RPiPlay"```
After the mirroring window is opened, run this command and click on the window. The window is renamed to "RPiPlay" and can then be selected from Zoom's screen share menu.

fduncanh, 2022-01-25:
@MichaelRenMR 

yes, the xdotool manual fix is an alternative.

What didn't work with the patch?