FD-/RPiPlay #225: Video cannot be reproduced from TV+ app

jcconca, 2021-02-26:
After installing the service, almost everything is fine:

iPhone XR mirroring
iPad 4 mirroring

iPhone photos, sometimes hangs video, get frezed and then recover and share it. But that is not the case.

I can even see the application youtube on full screen mirroring.

But when I try to reproduce TV+ in the rpiplay, the error says something like, the screen has not authorisation to reproduce protected films

[image]

If I disable the screen mirroring and try to send directly the video, I can see errors in the console saying:
aacDecoder_DecoderFrame error : 0x400c
aacDecoder_DecoderFrame error : 0x4002

[image]

No video or audio sent to the raspberry.

Is a known incompatibility with the application or is the content?

FD-, 2021-02-26:
I suspect the problem is issue #7.

closed 2021-02-26 by FD-

jcconca, 2021-02-26:
@FD- thanks for the quick reply.

I will have a look into the the related problem. I’m rusty with C, but if I can help I will try