FD-/RPiPlay #229: Multiple RpiPlay on the same Network

Q3F1S9B, 2021-03-13:
Hey, 

I'm looking for running multiples Rpiplay in a school. 
Each class has a PC with Debian Buster with a video projector. 
We wan't to use our iPads ( annotation, photo, ... ) and stream it on the projector. It work very well with your program, thanks.
 
However, when we launch rpiplay on two or more pcs  ( with different names ), only one appear on the control center. The first launched.

Sergi030, 2021-03-15:
Seems that this problem is caused by the fixed  DEFAULT_HW_ADDRESS on the rpiplay.cpp

I think the cool solution would be to calculate it at every host dynamically. The main problem would be to know which interface is the "correct" one. (also needs to be coded code for every different platform)

Another solutions are to launch with a random mac every time (now it's also not correct with the hosts, and it works).

Or we can add an optional new parameter to ask for the mac address directly.

A solution that you can implement directly is to compile the software changing DEFAULT_HW_ADDRESS for every pc that you want to use.

Sergi030, 2021-03-15:
I see that a rpiplay now tries to know the mac address assuming that would be eth0 or wlan0 on a linux system.

The solution of an optional parameter should be the logical with this new info

Sergi030, 2021-03-15:
I made this PR solving this: https://github.com/FD-/RPiPlay/pull/230/

Q3F1S9B, 2021-03-15:
Thank you ! 
I don't work today so I will test it tomorrow, will keep you updated !

clintinthecode, 2021-05-06:
Is there a solution to allow multiple instances on the same raspberry Pi? I have two monitors and currently have a windows machine running two 4k displays, and I'm using two different programs on windows, one for the airplay on the left monitor, another one for the right monitor. Looking to go to raspberry pi instead of a windows machine and found this repo that might do what I need.

Sergi030, 2021-05-08:
> Is there a solution to allow multiple instances on the same raspberry Pi? I have two monitors and currently have a windows machine running two 4k displays, and I'm using two different programs on [...]

I think if I add a random option for the DEFAULT_HW_ADDRESS you will be able to run multiple rpiplay.

Sergi030, 2021-05-08:
I added an option called --randomMac that allows you to run more than one rpiplay instances on the same pc. 

With this option a random mac will be calculated, and also it generates a random name for the rpiplay instance
`./rpiplay --randomMac `

Also if you want to set a custom name just simply use the -n option with the --randomMac
`./rpiplay --randomMac -n leftScreen`

Sergi030, 2021-05-08:
I made a new PR with the 2 changes.

https://github.com/FD-/RPiPlay/pull/244