FD-/RPiPlay #235: RPiPlay does not show up on iPhone

Anth0rx, 2021-03-22:
I have installed all dependencies and successfully built RPiPlay. When I run `./rpiplay` I am not able to see the instance on my iPhone. I am on the same network and I have ensured the iPhone is connected via 5 GHz. Is there anything I have missed?

nzohrab, 2021-03-24:
Running a VPN could cause this issue

Anth0rx, 2021-03-25:
No, a VPN is not running. Is there any other way I can troubleshoot this problem?

JSAnyone, 2021-03-30:
I have the same problem. When starting rpiplay with `rpiplay -b on -n Wohnzimmer` I get three warnings, but I think they are normal:
```
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
```

JSAnyone, 2021-03-30:
I just found out that it depends on the device I want to connect to RPiPlay:
An iPhone Xs with iOS 14.4 isn't working, while an iPad Air 2 with iOS 13.5.1 ist working, luckily I haven't updated this iPad yet.
Anyone else experiencing problems with iOS 14(.4)?

JSAnyone, 2021-04-12:
Since the newest iOS release it's working again. No idea why, but I'm happy, Thanks for this great project!

FD-, 2021-07-15:
Usually caused by some problem in your network.

closed 2021-07-15 by FD-