FD-/RPiPlay #237: Is there an official icon for rpiplay?

Botspot, 2021-03-24:
I'm interested in adding rpiplay to the [Pi-Apps RPi app store](https://github.com/Botspot/pi-apps).

When displayed in the app list, rpiplay will need an icon to go with it. Is there an official (or unofficial) icon that can be associated with rpiplay?

Thanks in advance.

CleanMachine1, 2021-04-09:
@FD- ?

FD-, 2021-04-09:
We don’t have an official icon yet I’m afraid. Any suggestions?

CleanMachine1, 2021-04-09:
First of all, I hope you are OK with Botspot and me adding RPi-Play.

I suggested the official logo provided by Apple, would you agree with that choice

FD-, 2021-04-12:
No, please do NOT use the official AirPlay icon since I’m sure you do not have Apple’s permission to do so.

CleanMachine1, 2021-04-12:
Ok thanks for your help.