FD-/RPiPlay #240: Decoding artifacts when using gstreamer

gamelaster, 2021-04-20:
[image]

OS: Ubuntu 20.04 (x86_64, virtual machine)
Phone: iPhone 7 (iOS 14)

For some reason, when using gstreamer, I have those artifacts. It looks like pixel format or width & height of received data is somehow broken. This also might be issue because of Virtual Machine, although, hard to say. Will investigate more later.

dougg3, 2021-04-29:
I actually ran into this problem as well and [wrote a blog post about it](https://www.downtowndougbrown.com/2021/04/finding-and-fixing-a-vmware-xorg-video-display-bug-using-gstreamer/). It’s a problem with VMware’s Xorg plugin, which is also used by VirtualBox in certain conditions. I submitted a patch but it hasn’t been accepted yet.

gamelaster, 2021-04-29:
@dougg3 wow, really amazing work finding out what is the issue! 😊 

Well, since it looks like it's Xorg issue, let's close this.

Thanks

closed 2021-04-29 by gamelaster

FD-, 2021-04-29:
Thanks @dougg3! Really impressive work you did there!

dougg3, 2021-04-29:
Thanks! My patch seems kind of stuck in limbo, I’m not sure if the VMware plugin has a maintainer right now. For now you could hack the code to use glimagesink instead of autovideosink and it should work better.