FD-/RPiPlay #243: Mirror-window does not open

Andr3wid, 2021-04-29:
Hi there!

Just built the latest version of RPIPlay on my Linux box (openSUSE Leap) and it is launching fine.
I connect to the PC from my iPad via the "Screen Mirroring"-button, the connection details get displayed as well but the corresponding "mirror-window" (window displaying the screen of my iPad) does not pop up on the PC. I installed all GStreamer-plugins as far as I am aware and I am left clueless...

This is some log-output when starting in debug-mode:

```
❯ ./rpiplay -d
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 28
Local: 192.168.0.170
Remote: 192.168.0.152
httpd receiving on socket 28
conn_request
Handling request GET with URL /info
INFO len = 345561092
httpd receiving on socket 28
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 28
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 28
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 28
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 28
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 28
conn_request
Handling request SETUP with URL rtsp://192.168.0.170/10206856039803469901
DACP-ID: 3C1E64F326949AD6
Active-Remote: 841040577
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 58981
raop_ntp parse remote ip = 192.168.0.152
raop_ntp starting time
raop_rtp parse remote ip = 192.168.0.152
raop_rtp_mirror parse remote ip = 192.168.0.152
eport = 35417, tport = 57825
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1619069162689044
httpd receiving on socket 28
conn_request
Handling request GET with URL /info
INFO len = 345561092
Accepted IPv4 client on socket 30
Local: 192.168.0.170
Remote: 192.168.0.152
httpd receiving on socket 28
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.0.170/10206856039803469901
httpd receiving on socket 28
conn_request
Handling request RECORD with URL rtsp://192.168.0.170/10206856039803469901
raop_handler_record
httpd receiving on socket 28
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.0.170/10206856039803469901
httpd receiving on socket 28
conn_request
Handling request SETUP with URL rtsp://192.168.0.170/10206856039803469901
DACP-ID: 3C1E64F326949AD6
Active-Remote: 841040577
Transport: null
type = 110
streamConnectionID = 1508118470772805564
raop_rtp_mirror starting mirroring
Mirroring initialized successfully
raop_rtp_mirror accepting client
raop_rtp_mirror width_source = 730.000000 height_source = 972.000000 width = 762.000000 height = 1016.000000
raop_rtp_mirror sps size = 17
raop_rtp_mirror pps size = 4
raop_rtp_mirror video ntp = 1619692514040624, now = 1619692514024190, latency = -16434
raop_rtp_mirror video ntp = 1619692514073909, now = 1619692514038833, latency = -35076
raop_rtp_mirror video ntp = 1619692514090570, now = 1619692514057154, latency = -33416
raop_rtp_mirror video ntp = 1619692514107228, now = 1619692514071635, latency = -35593

```

bendschs, 2021-06-15:
same here, with rpi renderer the window does not open., my log looks similar (no hint given, everything seems to be working but without seeing a window on the desktop).

cheers
bendsch

fduncanh, 2021-09-14:
@Andr3wid

if its gstreamer / X11

perhaps autovideosink is choosing something weird for the sink.

try editing video_renderer_gstreamer.c and change autovideosink to ximagesink or xvimagesink

heatherathebyne, 2021-10-05:
OpenSUSE Tumbleweed user here, experiencing the same issue. I tried `ximagesink` and `xvimagesink` without success.

heatherathebyne, 2021-10-05:
A little bit more investigation yielded the following: running `GST_DEBUG=4 ./rpiplay -vr gstreamer -ar dummy -d` with current `master`, in the GStreamer debug output it looks like the pipeline breaks after `decodebin`.

```
0:00:00.007335435   523      0x13d0200 INFO            GST_PIPELINE gst/parse/grammar.y:689:gst_parse_perform_link: linking some pad of GstDecodeBin named decodebin0 to some pad of GstVideoConvert named videoconvert0 (0/0) with caps "(NULL)"
0:00:00.007339405   523      0x13d0200 INFO        GST_ELEMENT_PADS gstutils.c:1816:gst_element_link_pads_full: trying to link element decodebin0:(any) to element videoconvert0:(any)
0:00:00.007344514   523      0x13d0200 INFO                GST_PADS gstpad.c:4314:gst_pad_peer_query:<videoconvert0:src> pad has no peer
0:00:00.007450494   523      0x13d0200 INFO        GST_ELEMENT_PADS gstelement.c:1006:gst_element_get_static_pad: no such pad 'src_%u' in element "decodebin0"
0:00:00.007459993   523      0x13d0200 INFO        GST_ELEMENT_PADS gstutils.c:1270:gst_element_get_compatible_pad:<decodebin0> Could not find a compatible pad to link to videoconvert0:sink
0:00:00.007464193   523      0x13d0200 INFO                 default gstutils.c:2206:gst_element_link_pads_filtered: Could not link pads: decodebin0:(null) - videoconvert0:(null)
```

Before today I knew next to nothing about GStreamer, so I could be off the mark on this, but https://stackoverflow.com/questions/28033813/unable-to-link-gstreamer-decodebin-to-jpegenc-in-application suggests that this could be due to `src_%u` in `decodebin` being only _sometimes_ present.

fduncanh, 2021-10-05:
Use UxPlay 1.37 http://github.com/FHD2/UxPlay   (RPiPlay fork)  with
./uxplay -vs ximagesink

fduncanh, 2021-10-05:
@heatherathebyne 
Try using UxPlay 1.37 http://github.com/FDH2/UxPlay   (RPiPlay fork)  with
./uxplay -vs ximagesink
.