FD-/RPiPlay #245: Issue with iOS >14.5

catchMyException, 2021-05-12:
_Originally posted by @DatNoHand in https://github.com/FD-/RPiPlay/issues/235#issuecomment-838197530_

I can confirm, that an iPhone with iOS 14.5.1 can't find the server either. Might be an issue with the latest iOS update. No error message is shown.

fduncanh, 2021-06-23:
works fine on IOS14.6 iphone SE (new)

RafaelGreenblatt, 2021-09-14:
I recently noticed that my iPad (iPad Air 2 with iOS 14.7.1), which had worked with RPiPlay in the past, no longer detects the server.  Oddly enough it works fine on my wife's iPad mini, which also runs iOS 14.7.1 (same computer, same network).  I'm quite puzzled.

fduncanh, 2021-09-14:
iPad 7th generation works with 14.7.1
about to upgrade to 14.8 .....
hoping for the best