FD-/RPiPlay #246: Unable to find rpiplay with nixos build

kolaente, 2021-05-31:
I'm trying to build RPiPlay for nixos. Building works, but my ios device is unable to find the RPiPlay service running on my pc.

I would guess this has something to do with either dns or some other discovery mechanism. Any idea how to debug this?

This is the `default.nix` file to reproduce:

```nix
{ stdenv, lib, fetchgit, cmake, openssl, avahi-compat, libplist, gst_all_1, pkg-config, pcre, libunwind, elfutils, avfs }:

stdenv.mkDerivation rec {
  name = "RPiPlay";

  src = fetchgit {
    url = "https://github.com/FD-/RPiPlay.git";
    rev = "5611a45aff8a3f818a6e9de3071577b1affab06d";
    sha256 = "063jnd9fwypqkpb5h1z1frizxlc9hd42iclvv5w3998j2rh4m3r0";
  };

  buildInputs = [ 
    cmake 
    avfs 
    elfutils 
    libunwind.dev 
    pcre.dev 
    pkg-config 
    openssl 
    avahi-compat 
    libplist 
    gst_all_1.gstreamer gst_all_1.gst-plugins-bad gst_all_1.gst-plugins-base gst_all_1.gst-libav gst_all_1.gst-vaapi
  ];

  runtimeDependencies = [ 
    avfs 
    elfutils 
    libunwind.dev 
    pcre.dev 
    pkg-config 
    openssl 
    avahi-compat 
    libplist 
    gst_all_1.gstreamer gst_all_1.gst-plugins-bad gst_all_1.gst-plugins-base gst_all_1.gst-libav gst_all_1.gst-vaapi
  ];
}
```

kolaente, 2021-06-03:
I've figured out I need a running avahi daemon with nssmdns enabled and user services publishing enabled to get my ipad to find the pc.

Now I'm hitting https://github.com/FD-/RPiPlay/issues/195

sherubthakur, 2021-07-28:
@kolaente Can you share what you did on NixOS that got it wokring? (if it is working on NixOS - my assumption is that you have it working on NixOS but not on ubuntu).

I have the following config.

```nix
  services.avahi = {
    enable = true;
    nssmdns = true;
    publish.enable = true;
    publish.userServices = true;
  };
```
I am using the derivation that you provide here. and on executing `rpiplay` I get the following output/warnings/errors
```shell
~ took 9s 
❯ rpiplay

(rpiplay:93195): GLib-GObject-WARNING **: 17:53:28.399: invalid cast from 'GstQueue' to 'GstBin'

(rpiplay:93195): GStreamer-CRITICAL **: 17:53:28.399: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(rpiplay:93195): GLib-GObject-WARNING **: 17:53:28.399: invalid cast from 'GstQueue' to 'GstBin'

(rpiplay:93195): GStreamer-CRITICAL **: 17:53:28.399: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(rpiplay:93195): GLib-GObject-WARNING **: 17:53:28.399: invalid cast from 'GstQueue' to 'GstBin'

(rpiplay:93195): GStreamer-CRITICAL **: 17:53:28.399: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(rpiplay:93195): GLib-GObject-WARNING **: 17:53:28.399: invalid cast from 'GstQueue' to 'GstBin'

(rpiplay:93195): GStreamer-CRITICAL **: 17:53:28.399: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(rpiplay:93195): GLib-GObject-CRITICAL **: 17:53:28.399: g_object_set: assertion 'G_IS_OBJECT (object)' failed
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
```

Other info:
- I am not running this on a RasberyPi (My assumption is that it should just work on a normal pc too) Is this assumption correct?
- I don't have access to a 5GHz bad wifi I am connected to a 2.4GHz wifi, I am not sure if this impacts it's functioning or not, as I read somewhere that it does.
- Tried it with firewall on and off.

kolaente, 2021-07-28:
The avahi config is the same I have.

> Can you share what you did on NixOS that got it wokring?

I got it working insofar as I was able to select the rpiplay server running on my nixos system as a target from the Ipad. I did _not_ get a window with the streaming output, similar to #195.

> my assumption is that you have it working on NixOS but not on ubuntu

It works fine in my ubuntu vm when setting it up with the instructions in the readme of this repo. Being ubuntu though, I don't really know what services are installed by default on it which may be required by rpiplay.

> I am not running this on a RasberyPi (My assumption is that it should just work on a normal pc too) Is this assumption correct?

Jup, as I said, running fine on ubuntu.

> I get the following output/warnings/errors

Same, afaiu they don't seem to affect running it. But then again, it isn't really working so maybe that does something.