FD-/RPiPlay #247: No audio/sound through HDMI

majuss, 2021-06-04:
Latest raspbian on Pi 4/2 GB

So I started RPIplay in the following manner: `rpiplay -a hdmi` but I can't get any sounds to work. The output on my Mac is set correctly, a alsa test command is playing back audio on my TV.

The logs show:
```
raop_rtp starting audio
RAOP initialized success
```

MiracleXYZ, 2021-07-16:
I came through this issue yesterday, but managed to fix the audio thanks to [this reply](https://github.com/FD-/RPiPlay/issues/176#issuecomment-767706910).

Just follow the build steps for Ubuntu 18.04/20.04 and install `gstreamer1.0-tools` and `gstreamer1.0-pulseaudio`:

```
sudo apt-get install cmake libavahi-compat-libdnssd-dev libplist-dev libssl-dev \
    libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-libav \
    gstreamer1.0-vaapi gstreamer1.0-plugins-bad
sudo apt-get install gstreamer1.0-tools gstreamer1.0-pulseaudio
```

Then rebuild and install the binary:

```
mkdir build
cd build
cmake ..
make
sudo make install
```

I ran the following command (using `-ar gstreamer`) and audio worked like charm.

```
rpiplay -n MyRpiPlay -a hdmi -ar gstreamer -l
```

RussQuan, 2021-08-20:
> 我昨天遇到了這個問題，但由於[此回复](https://github.com/FD-/RPiPlay/issues/176#issuecomment-767706910)，我設法修復了音頻。
> 
> 只需按照 Ubuntu 18.04/20.04 的安裝步驟進行安裝`gstreamer1.0-tools`和`gstreamer1.0-pulseaudio`：
> 
> ```
> sudo apt-get install cmake libavahi-compat-libdnssd-dev libplist-dev libssl-dev \
>     libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-libav \
>     gstreamer1.0-vaapi gstreamer1.0-plugins-bad
> sudo apt-get install gstreamer1.0-tools gstreamer1.0-pulseaudio
> ```
> 
> 然後重建並安裝二進製文件：
> 
> ```
> mkdir build
> cd build
> cmake ..
> make
> sudo make install
> ```
> 
> 我運行了以下命令（使用`-ar gstreamer`），音頻效果很好。
> 
> ```
> rpiplay -n MyRpiPlay -a hdmi -ar gstreamer -l
> ```

This works sooooo good !! Thank you!

closed 2021-12-14 by majuss