FD-/RPiPlay #250: Documentation suggestion(readMe.md)

jonathanwodzisz, 2021-06-18:
make -j causes the pi to crash, I would suggest not using -j because pi's are slow and will crash when the cpu is working too hard. Not making in parallel is slower but worth noting in the documentation.

FD-, 2021-07-04:
No Pi should crash when the cpu is working too hard. That's more likely a problem caused by your PSU.

jonathanwodzisz, 2021-07-05:
> No Pi should crash when the cpu is working too hard. That's more likely a problem caused by your PSU.

Yes sir, that's exactly why, most Pi owners don't have adequate power which causes their Pi to crash, just like mine did. Power supplies for Pis are often improvised.

jonathanwodzisz, 2021-07-05:
> No Pi should crash when the cpu is working too hard. That's more likely a problem caused by your PSU.

There's a reason my request has a a few likes,they had the same issue, on a Pi you should not use the cpu to its full capacity in general when it's being used by average Pi users. It sucks to have to make the sacrifice, I assume you aren't using this code anymore yourself so I would personally not put in the documentation to use -j when not necessary. The project is small enough to not need it anyways.