FD-/RPiPlay #253: Cannot build - /usr/bin/ld: /opt/vc/lib/libbrcmGLESv2.so: error adding symbols: file in wrong format

vantino, 2021-07-15:
Ciao

I am running a model 4
Linux pi4 5.10.46-v8+ #1433 SMP PREEMPT Mon Jul 5 15:04:57 BST 2021 aarch64 GNU/Linux

Followed your installation instructions, but cannot build. Get the following msg:

pi@pi4(10b):~/src/RPiPlay/build $ make -j
[  5%] Built target playfair
[  7%] Built target llhttp
[ 11%] Built target h264-bitstream
[ 13%] Built target ilclient
[ 59%] Built target airplay
[ 95%] Built target fdk-aac
[ 98%] Built target renderers
[ 99%] Linking CXX executable rpiplay
/usr/bin/ld: /opt/vc/lib/libbrcmGLESv2.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rpiplay.dir/build.make:99: rpiplay] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:208: CMakeFiles/rpiplay.dir/all] Fehler 2
make: *** [Makefile:130: all] Fehler 2

Any suggestions, idea, help .. ?

THanks
Vantino

FD-, 2021-07-15:
64bit OSs are not supported, since there's not compatible version of OpenMAX. See #166

closed 2021-07-15 by FD-

vantino, 2021-07-15:
ok - thanks

RodargDev, 2022-02-06:
> 64bit OSs are not supported, since there's not compatible version of OpenMAX. See #166

Any chance 64 bit will be supported in the future or? Maybe put it in the readme if not because people like myself try to get this working and spend a lot of time doing so just to stumble upon this answer and feel like a lot of time has been wasted

LockonS, 2022-07-04:
Same here, since 64 bit os is not supported yet, add that into readme might save other people a lot of time.