FD-/RPiPlay #256: Issues when viewing photos full screen in IPadOS Photos app

laolux, 2021-07-28:
Trying to view photos fullscreen on the ipad in the photos app works fine, but does not properly get displayed on my computer by rpiplay.
The picture I try to view is 1392x972 pixel in size and is a blue/red checker board with a one pixel grid. Click on the picture to see it for real, the preview in github is a bit strange.
[image]
This is how everything looks like when viewing on the ipad (screenshot done on ipad):
[image]
This is how it looks like when viewing on my computer via rpiplay:
[image]
Clearly something is broken here. When I switch back to the picture overview, then everything is displayed correctly again in rpiplay.

FD-, 2021-07-28:
What exactly is clearly broken here? The photos app directly renders to the second (AirPlay) screen, so don’t expect a direct copy of what you see on the iPad screen. What does a real AppleTV display here?

laolux, 2021-07-29:
First I need to correct myself. The issue appears with pretty much any picture, not just the example above.

What looks broken to me are the black borders around the picture. Top/bottom I can understand, as the size of the picture does not match the size of the display, but left/right? Also, when I open a screenshot from my ipad (which automatically has the correct size), then I get huge black borders everywhere, see attached picture below. And those borders do not appear when I open the overview of all my pictures.

Unfortunately I cannot compare with a real AppleTV, as I do not own one. Maybe I can find a friend who has one, so I could check if those huge black borders are there as well and thus expected.

[image]

fduncanh, 2021-08-07:
are you using an added feature to set the display size?
This isnt yet part of  "official" RPiPlay.

If you want it on RPiPlay, it is also in my pull request #257
or (for gstreamer users only), UxPlay 1.3 at  https://github.com/FHD2/UxPlay.git 

From what I see, usually the image uses as much of the display area that its aspect ration allows, by sometimes is smaller, leaving black areas on both top and bottom and right and left, as in your      The settings (default 1920x1080) are (I think) used as a starting point for negotiation with the iPad client.    By adjusting the wxh aspect ratio, I could Photos images to fill the display window with a tiny black edge, on a standard  iPad (I think you are using a pro).   Maybe if you experiment with the -x -y setting  on PR #191 or the -s wxh setting on #257 or UxPlay 1.3 you can find one that is good for iPad pro.

laolux, 2021-08-11:
No, this happens when using the latest "official" version, without trying to set the resolution by hand.

I just tried zooming into the picture -> I can make the black border left and right disappear, but not top/bottom. This works regardless whether I am using portrait or landscape mode. I also noticed that the pictures always seem to be shown in landscape mode, even when being displayed in portrait mode on my ipad.

Furthermore, I just tried with an iphone. Result is the same is with ipadPro.

Thanks for the link to a new repo for UxPlay, I will check it out. (link has a typo, FHD2 instead of FDH2)