FD-/RPiPlay #261: Mirroring on dual monitor RPi 4

DevPizzini, 2021-08-04:
I am using a Raspberry Pi 4 with 2 monitors attached and when I launch RpiPlay it only uses the main one.
It would be great to be able to use both of them for mirroring.

Thanks in advance

fduncanh, 2021-08-12:
this is an X11 xwindows issue  rpiplay makes no decision about window placement.

rpiplay -> gstreamer -> X11

Install  xdotool if you don't have it, I think it  can be used to move windows between  different screens.

if it is not in your distro,  find it  on github.

Bartiff, 2021-08-13:
@DevPizzini : Did you manage to do this? I'm interested in that too.

mw2323, 2022-03-26:
Also interested in this. I have a pi4 running official 7" touchscreen, and want to use RPiPlay to output to a separate HDMI-fed display. Running Buster lite so no GUI. RPiPlay seems to only display on the small touchscreen, rather than the huge tv it's connected to.