FD-/RPiPlay #262: Change Readme to „make -j1“ to prevent crashing raspberry

crustulumtheoneandonly, 2021-08-09:
To prevent the Raspberry Pi (3Bplus) from crashing when building over ssh, use the Command „make -j1“. Please change this in the Readme file under:

For building on a fresh Raspbian Stretch or Buster install, these steps should be run:
.
.
.
.
.
make -j1

RuBiCK, 2021-08-17:
I had the same issue. Compiling with `-j1` almost finished when it crashes.

I'm using RPI Zero W 

``` 
Scanning dependencies of target rpiplay
[ 99%] Building CXX object CMakeFiles/rpiplay.dir/rpiplay.cpp.o
[100%] Linking CXX executable rpiplay
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_start':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:280: undefined reference to `ilclient_change_component_state'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:281: undefined reference to `ilclient_enable_port_buffers'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:282: undefined reference to `ilclient_change_component_state'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:277: undefined reference to `ilclient_change_component_state'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_set_volume':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:371: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:371: undefined reference to `ilclient_get_handle'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_render_buffer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:349: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:349: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:331: undefined reference to `ilclient_get_input_buffer'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_destroy_renderer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:98: undefined reference to `ilclient_disable_tunnel'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:99: undefined reference to `ilclient_disable_port_buffers'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:100: undefined reference to `ilclient_teardown_tunnels'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:102: undefined reference to `ilclient_state_transition'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:103: undefined reference to `ilclient_state_transition'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:104: undefined reference to `ilclient_cleanup_components'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:108: undefined reference to `ilclient_destroy'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_init_renderer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:129: undefined reference to `ilclient_create_component'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:145: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:145: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:178: undefined reference to `ilclient_setup_tunnel'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:191: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:191: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:212: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:212: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:227: undefined reference to `ilclient_get_handle'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o):/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:227: more undefined references to `ilclient_get_handle' follow
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_init_renderer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:119: undefined reference to `ilclient_init'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_destroy_renderer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:98: undefined reference to `ilclient_disable_tunnel'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:99: undefined reference to `ilclient_disable_port_buffers'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:100: undefined reference to `ilclient_teardown_tunnels'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:102: undefined reference to `ilclient_state_transition'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:103: undefined reference to `ilclient_state_transition'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:104: undefined reference to `ilclient_cleanup_components'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:108: undefined reference to `ilclient_destroy'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_init_renderer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:153: undefined reference to `ilclient_create_component'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:167: undefined reference to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:167: undefined reference to `ilclient_get_handle'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audio_renderer_rpi_destroy_renderer':
[... cut ...]
/usr/bin/ld: /home/pi/RPiPlay/renderers/video_renderer_rpi.c:139: undefined reference to `ilclient_disable_tunnel'
/usr/bin/ld: /home/pi/RPiPlay/renderers/video_renderer_rpi.c:137: undefined reference to `ilclient_disable_tunnel'
/usr/bin/ld: /home/pi/RPiPlay/renderers/video_renderer_rpi.c:138: undefined reference to `ilclient_disable_tunnel'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rpiplay.dir/build.make:99: rpiplay] Error 1
make[1]: *** [CMakeFiles/Makefile2:208: CMakeFiles/rpiplay.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

KDKHD, 2022-09-21:
Same as @RuBiCK

Matheo5, 2023-01-29:
So did someone found an issue ?

visit1985, 2024-04-17:
In my case it turned out, that `renderers/CMakeFiles/ilclient.dir/opt/vc/src/hello_pi/libs/ilclient/ilclient.c.o` got corrupted (0 bytes) during the crash and make didn't detect that during subsequent builds. Deleting this file and running `make -j1` again finished successful.