FD-/RPiPlay #265: Latency gradually increases; video is slower than device's screen?

notkirito, 2021-08-18:
It appears that the latency of the video feed gradually increases. However, in close inspection, it actually appears that the rendered video feed is slower than the device's screen, which would explain that the latency gradually increases.
```
+-------+-----------------------------------------------------------------------------------------+
|Device | frame1 frame2 frame3 frame4 frame5 frame6 frame7 frame8 frame9 frame10                  |
+-------+-----------------------------------------------------------------------------------------+
|RPiPlay| frame1   frame2   frame3   frame4   frame5   frame6   frame7   frame8   frame9   frame10|
+-------+-----------------------------------------------------------------------------------------+
```
I can't explain this very well, I hope you find this table helpful.

pallas, 2021-08-19:
Are you using gstreamer or ilclient?

notkirito, 2021-08-19:
Unknown, If it helps, I use this command-line: `rpiplay -n rpiplay_72849872 -l -a off`

notkirito, 2021-08-22:
I think it's gstreamer.

notkirito, 2021-08-23:
Is this known to happen when the GPU is weak? I have an Intel "HD" Graphics iGPU.

fduncanh, 2021-09-25:
If you are not using the Raspberry Pi (just desktop linux),   you could test on the RPiPlay derivative UxPlay http://github.com/FDH2/UxPlay.   Although this has some improvements,  you will probably find the same issue.  It would be useful to know,.

notkirito, 2021-10-05:
Unfortunately, I have the same issue there. Sorry for late reply, I do not have much free time at the moment.

GRrHaG, 2022-04-23:
i have a same problem with RPI 3 buster 32b, do you have an issues ?

dabreaka, 2023-04-07:
I have the same problem with a RPI 3 B - 1 GB main memory ...

fduncanh, 2023-04-07:
If you are willing to update from Buster to RPi OS Bullseye,  UxPlay1.63  https://github.com/FDH2/UxPlay   (derived from RPiPlay, but now greatly evolved) may solve your issue  (use the -vsync option of UxPlay-1.63).  Its RPi support is developed on RPi 4B, but is reported to work on R Pi 3B.     Test it on a spare micro SD card.