FD-/RPiPlay #267: iPhone can show 'RPiPlay' services in mirroring button, but it doesn't work on the pi.

xmasrain, 2021-09-19:
[image]
[image]

it seems that rpiplay didnt receive any requests from the iPhone.

fduncanh, 2021-09-20:
If the server was a Linux PC, I would say for sure that you are blocked by a firewall on the server.      If the pi has a firewall, switch it off to see if that helps.   Possible firewalls that might be running are are ufw and firewalld   The avahi "WARNING" is completely harmless.

Badrmoh, 2021-11-10:
same problem on pi4 and pi zero 2 and no firewall is configured

BPetty-InfoSec, 2021-11-13:
Tried with a Zero and a 4, no firewall, same issue.

lycheeEng, 2021-11-20:
Same problem here, I use rpi2, no firewall, and the pi can play audio files.