FD-/RPiPlay #271: Can't connect from macos

luis901101, 2021-09-26:
Hi, first of all thanks for this awesome tool.
Currently I'm able to connect to the RPiPlay using my iPhone 6s and my iPad 6th gen, both with iOS 14.8 and iPad 14.8 respectively. But my Macbook Pro 15" mid 2015 with MacOS Big Suer 11.6 can't connect.
I tried with some forks that allows to specify custom screen resolution and frame rate, but still can't connect.
The error message says:
> Can't connect to "RPiPlay"

and `rpiplay -d` only shows the stream info channel, sample rate, frame size, aot and bitrate.

fduncanh, 2021-09-27:
UxPlay 1.35 https://github.com/FDH2/UxPlay is a (GStreamer-only)  RPiPlay fork (intended for Desktop Linux etc, not RPi) that definitely works with macOS clients (this is verified  on unibody MacBook Pro on Catalina 10.15).  You could test with that to see if its a Big Sur issue.

luis901101, 2021-09-27:
Hi @fduncanh thanks for your anawer, but what I need is to use RPi as AirPlay server for all my devices, I need to be able to connect my macbook to the RPiPlay server. I don’t intend to use my macbook as an AirPlay server.

fduncanh, 2021-09-28:
UxPlay 1.35 running on Linux(or MacOS)  accepts macOS clients for screen sharing.  It hasn't been tested with a Big Sur client, but does mirror a unibody MacBook Pro running Catalina.   Since its basically RPiPlay with GStreamer renderers only, plus enhancements, you could test it on a Linux desktop (or maybe even a RPi) to check that macOS 11 Big Sur still supports legacy Airplay  (its possible Apple removed it from Big Sur?).  This would be useful to know, and helpful for getting Macbook mirroring on RPiPlay to work, it Apple didnt do this.

fduncanh, 2021-09-28:
@luis901101

UxPlay 1.35 is a linux(and MacOS) Airplay server that mirrors macOS Catalina 10.15.x clients.  It's not tested with Big Sur clients.
It will work on a Linux desktop, and maybe RPi (Gstreamer renderers only).   If you test, you can see if Big Sur clients still work (Its always possible that Apple removed Legacy AIrPlay protocol from Big Sur).

If it works on UxPlay 1.35, it can be made to work on RPiPlay, since UxPlay 1.35 is current RPiPlay plus enhancements, but is GStreamer only.

FD-, 2021-09-28:
This is very likely a network problem.

fduncanh, 2021-09-28:
@luis901101 

I just verified that RPiPlay (or at least the version in my PR #266 )  (plus PR #263 but this shouldnt matter)
using GStreamer renderers  does mirror a Macbook Pro running macOS 10.15.7 Catalina,  It's up to you to test with a  Big Sur client.

luis901101, 2021-09-28:
Ok, thanks, I will test it later.

luis901101, 2021-10-02:
I tested it in another Macbook Pro (a 16 inch 2019 model) with the same MacOS BigSur version 11.6 and it worked fine... No idea what happen with the other macbook... anyway rpiplay works fine.

closed 2021-10-02 by luis901101