FD-/RPiPlay #273: make fails

spicemint, 2021-10-04:
installing on ubuntu 16.04, libplist-dev is installed
Linux SAT 4.4.0-210-generic #242-Ubuntu SMP Fri Apr 16 09:57:56 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

`make
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenMAX libraries not found, skipping compilation of Raspberry Pi renderer
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jody/RPiPlay/build
[ 17%] Built target playfair
[ 28%] Built target llhttp
[ 80%] Built target airplay
[ 94%] Built target renderers
Scanning dependencies of target rpiplay
[ 97%] Linking CXX executable rpiplay
lib/libairplay.a(raop.c.o): In function `raop_handler_setup':
raop.c:(.text+0x7c8): undefined reference to `PLIST_IS_DATA'
raop.c:(.text+0x7da): undefined reference to `PLIST_IS_ARRAY'
raop.c:(.text+0xb26): undefined reference to `PLIST_IS_DATA'
collect2: error: ld returned 1 exit status
CMakeFiles/rpiplay.dir/build.make:100: recipe for target 'rpiplay' failed
make[2]: *** [rpiplay] Error 1
CMakeFiles/Makefile2:70: recipe for target 'CMakeFiles/rpiplay.dir/all' failed
make[1]: *** [CMakeFiles/rpiplay.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`
what else can I try?

fduncanh, 2021-10-04:
did you follow instructions?

sudo apt install libplist-dev

seems to be missing.....

fduncanh, 2021-10-04:
since you think you installed, but it seems to be missing, reinstall it.

fduncanh, 2021-10-04:
If you are not using RPi, you might try  the RPiPlay fork http://github.com/FDH2/UxPlay which behaves more correctly on desktop Linux

spicemint, 2021-10-05:
> since you think you installed libplist-dev, but it seems to be missing, reinstall it.
had tried it already but did it again, maybe it needs to be a different version?

sudo apt install libplist-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libplist-dev is already the newest version (1.12-3.1ubuntu0.16.04.1).

spicemint, 2021-10-05:
> If you are not using RPi, you might try the RPiPlay fork [http://github.com/FDH2/UxPlay](https://github.com/FDH2/UxPlay) which behaves more correctly on desktop Linux

thank you, just tried that one and gives similar error:

[100%] Linking CXX executable uxplay
lib/libairplay.a(raop.c.o): In function `raop_handler_setup':
/home/jody/UxPlay/lib/raop_handlers.h:349: undefined reference to `PLIST_IS_DATA'
/home/jody/UxPlay/lib/raop_handlers.h:397: undefined reference to `PLIST_IS_ARRAY'
/home/jody/UxPlay/lib/raop_handlers.h:349: undefined reference to `PLIST_IS_DATA'
collect2: error: ld returned 1 exit status
CMakeFiles/uxplay.dir/build.make:100: recipe for target 'uxplay' failed
make[2]: *** [uxplay] Error 1
CMakeFiles/Makefile2:70: recipe for target 'CMakeFiles/uxplay.dir/all' failed
make[1]: *** [CMakeFiles/uxplay.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

alexmyczko, 2021-10-06:
ubuntu 16.04 is pretty old, maybe try with 18.04 or 20.04?

closed 2021-10-06 by spicemint