FD-/RPiPlay #283: Possible aes ctr decryption bug in rpiplay ?

fduncanh, 2021-11-02:
I've been trying to get ALAC streams to play in rpiplay (and uxplay), with gstreamer.     Comparing the audio stream from shairplay (parent of raop in rpiplay) with that from rpiplay, I found that the last 16 entrypted bytes are each set to 00 (before any residual less than 16 unencrypted bytes are appended to the packet)     The same is true for the AAC-ELD mpeg v4 stream that does play in current  rpiplay.

This is very surprising:  is this a feature of an AAC-ELD packet?   (it this was a corruption of the AAC packet I wouldn't expect it to play, but perhaps AAC is resilent against losing those 16 bytes).

example of a decrypted (not yet decoded) AAC packet
```
8d867f5ffff056c625098e2241815f9d
773a132798b7d50262652a6160b24843
83e0d9ec841c7fff27f01308a20a1f36
8a0759aaafb8771158d8fd04e2fd23f1
bf32b51c35a6fe5bf3badaf074ff6d4b
4f2ceafd3ce3639784759ce322a24637
252042b2cea79099c46a29c3445963f0
4a48f4074773176c10a3d23ad6dd4607
ae1ee145438880af13edf5f7e652b8ce
13db43265d183c49c19dae12a24c1994
5c2d18fdfe03d7de2b96adc1b553ba33
fc73831ba7c49acb6450720ea33e60e5
a41ad6926b4ceea8c41ecf60c4dfb74a
cde510cf15134409699db64d23f9f7ab
00000000000000000000000000000000                           <--------   unexpected!  (but audio still  plays)
927bf5aa5a5cab2d7f585aa800
```

example of a decrypted  (not yet decoded) ALAC packet  : (
```
2000000400130805ddfa0c0531fc4413
080369fd240363fdefff00067a3cd609
39bb74db4b2608120698035792da85e4
439510ae2ad18e336a2684921a01d3da
bc2249996b132db975a647230ab2aca9
4e260189225693b455011a54534e3df2
901193394d3a5d0a92cb2a391a55ade6
ab70b7516ea95e3b8c22a9298cdd21c4
98b01d8e7496ed2be914f12c68545415
7aace55197ba862241892a94c2931d5c
25a6836e9ba50a40bfe000ed66698c54
145087695086914dcb23aa4e2b70424d
a6e48bbd516c3536ac453da4ef0534f1
535b6d0c8eee9b6f39a0d42b520b74e2
1e6356921b7109d4a14057126535b34e
aa0cd31e5ba447b288442599549da451
8e9f71b1dd12d276da54a875954aa2c2
34aa32293114dc594d2c653ca834ba80
00000000000000000000000000000000                 <----------------- definitely an error (by comparision with shairplay)
65d48c5b6a9d2bac4d1c
```
The rest of the ALAC packet is good.

FD-, 2021-11-06:
This is an excellent find! Thanks for your efforts! I'm sorry I'm so slow with my responses at the moment, I'm just very busy with other stuff. If you submit a PR with just this fix (or a rework to make our crypto code follow the OpenSSL EVP guidelines), I'll gladly accept it :)

pallas, 2021-11-11:
I have a potential fix for this that involves using Final. I believe the issue is a padding setting, but I haven't tested with OpenSSL 3.0 yet & I'm a little afraid to install it on Raspberry Pi OS.

pallas, 2021-11-11:
Please check out the commit I just pushed, b9e096354fce474d5c9ba1e4a04a85bb9f8cdb03

pallas, 2021-11-15:
It sounds like that commit fixes this issue. I want to refactor the lib/crypto stuff into the actual callers & replace the `aes_init`/`aes_destroy` calls with calls to `EVP_DecryptInit_ex` once and `EVP_CIPHER_CTX_reset` to prevent a much of `malloc`/`free` in this tight loop.

closed 2021-11-15 by pallas