FD-/RPiPlay #293: Server work but no screen on my pi - wrong hdmi channel?

schmichris2, 2021-11-16:
I have the problem that after installing and starting the server, I can connect to my MacBook and the connection is established, but nothing is displayed on the Pi. There does nothing at all no black screen etc.. Is there possibly a problem with the output channel? It seems to me that the signal is not output to the correct HDMI port. Can this be? On which channel is the signal sent HDMI0 or HDMI1? Is it possible to change it? Thanks a lot!

BastelKlug, 2021-11-18:
same problem here 👋

Illaise, 2021-11-24:
Same problem, after rpiplay start, appears HDMI "No signal"

FD-, 2021-11-24:
> Same problem, after rpiplay start, appears HDMI "No signal"

I'm not sure this is the same issue as the others are reporting. @schmichris2 mentioned "no black screen".

@schmichris2's reported issue seems to be identical to #296. I'm closing this one because we already have more comments on the other issue.

closed 2021-11-24 by FD-