FD-/RPiPlay #294: New Raspberry Pi Renderers

FD-, 2021-11-18:
Raspberry Pi OS Bullseye removes OpenMAX, which RPiPlay uses for video and audio rendering on Raspberry Pi. Assuming the gstreamer renderers are not performant enough for the Pi Zero, we need:

 -  A video renderer based on V4L2 or MMAL (which apparently is still supported and will eventually be supported by 64-bit builds of Raspberry Pi OS.
 -  An audio renderer based on ALSA

pallas, 2021-11-18:
I was able to install the firmware package via rpi-update, I think they might have just removed the main package and moved it to rpi-update. I've been running RPiPlay on Bullseye for a few weeks now on a 3B+

8bignic8, 2021-11-21:
https://askubuntu.com/questions/384059/error-compiling-a-package-configure-error-no-gstreamer-1-0-1-0-0

sudo apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev

maybe?

arnlen, 2021-11-25:
> I was able to install the firmware package via rpi-update, I think they might have just removed the main package and moved it to rpi-update. I've been running RPiPlay on Bullseye for a few weeks now on a 3B+

Could you please explain what you've done step by step please?
I've done a rpi-update + gstreamer installation but RPiPlay still doesn't work: I'm stuck with an error related to `libopenmaxil.so` (see https://github.com/FD-/RPiPlay/issues/296#issuecomment-979482366).

a7mad3bdullah, 2023-09-27:
The only way I was able to make it work is by installing it on Raspberry Pi OS Buster Not Bullseye, see [364](https://github.com/FD-/RPiPlay/issues/364)