FD-/RPiPlay #295: Undefined symbol: gst_buffer_get_audio_level_meta

luminarycrush, 2021-11-20:
Fedora 33/latest patch level.  Builds fine, one warning (/export/home/jas/work/RPiPlay/renderers/video_renderer_gstreamer.c:33:17: warning: ‘check_plugins’ defined but not used [-Wunused-function]).

I go to run RPiPlay and get:

`(rpiplay:1317764): GStreamer-WARNING **: 15:24:52.373: Failed to load plugin '/usr/lib64/gstreamer-1.0/libgstlevel.so': /usr/lib64/gstreamer-1.0/libgstlevel.so: undefined symbol: gst_buffer_get_audio_level_meta
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
`
After this it appears to start/run OK, I can find it exploring for AirPlay devices.  

`Accepted IPv4 client on socket 18
Local: 192.168.100.3
Remote: 192.168.100.228
`

When I attempt to screenplay from an iPhone control center it appears to begin but then pauses after a few seconds.  I don't see any other evidence of why it's failing.

Next I try streaming audio only from a podcast and this time the podcast thinks it's playing but I get no audio (checked the mixer for any zeroed inputs, all were turned up):

raop_rtp starting audio
Got metadata of 82 bytes
Got metadata of 189 bytes
Got image data of 144892 bytes
Got metadata of 157 bytes
raop_rtp starting audio
Got image data of 144892 bytes
Got metadata of 198 bytes

I then try Youtube, similar to above.  Not sure where to troubleshoot or if there is some gstreamer library issue per the warning above.

fduncanh, 2021-11-27:
You seem to be streaming AirPlay (ALAC) audio  podcast which is not supported on RPiPlay.   Use -d debug option to see why
iPhone isnt mirroring.    (or try the [UxPlay](http://github.com/FDH2/UxPlay) derivative of RPiPlay)

pallas, 2022-02-25:
stale

closed 2022-02-25 by pallas