FD-/RPiPlay #296: Nothing happens after starting server

giansegato, 2021-11-21:
I’m quite sure this is something trivial, but I can’t get out of it.

I have a Raspberry Pi 3+ with a fresh Raspberry Pi OS Lite installed. If I plug an HDMI cable to the Pi, the CLI fires up on screen, as Lite doesn’t have any GUI. So far so good.

I successfully installed RPiPlay following the README instructions. I have it running via an `ssh` connection, in a `tmux` pane to avoid the command to stop after I drop the connection. The command is `rpiplay -n “TV” -a hdmi`. Everything runs smoothly, and I start seeing the server on both my iPad and iPhone.

If I select it as an AirPlay target to stream anything (be a video or screen mirroring), nothing happens on the monitor to which the Pi is connected (via HDMI). I just keep seeing the CLI. I saw in other issues that Lite is a supported OS, so I cannot understand what I’m missing.

I also tried to plug a keyboard directly to the board and run the command from there, but - still - nothing happens: only debug logs and the CLI itself.

What am I missing?

abdullahaltun70, 2021-11-22:
Same issue here. I have the excact same problem. This is the output I am getting as I'm sure you're getting as well:

pi@raspberrypi2:~/RPiPlay/build $ ./rpiplay -n Monitor -a hdmi
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 14
Local: 192.168.2.34
Remote: 192.168.2.3
Accepted IPv4 client on socket 16
Local: 192.168.2.34
Remote: 192.168.2.3
raop_rtp_mirror starting mirroring

FD-, 2021-11-22:
Are you using Bullseye?

abdullahaltun70, 2021-11-22:
> Are you using Bullseye?

Yes I am. I did a fresh install of rp os lite this morning. and ofcource apt update and upgrade .

FD-, 2021-11-22:
Please try the newest Buster image instead.

abdullahaltun70, 2021-11-22:
How do I properly install Buster? And where can i download it from. I don't know what buster is😅

brechtknecht, 2021-11-22:
I have exactly the same result and messages on a clean `Ubuntu 21.10 ubuntu tty1` like @abdullahaltun70.

FD-, 2021-11-22:
That suggests gstreamer is not properly configured on your system.

abdullahaltun70, 2021-11-22:
> 

How can I fix gstreamer not working?

FD-, 2021-11-22:
My suggestion would be reading through other similar issues that were eventually resolved. Gstreamer support was written by contributors, so maybe they can help better here.

abdullahaltun70, 2021-11-22:
I installed the Buster image and gstreamer (again). I followed a  few threads that where suggesting I should install all of the dependencies again which are shown in the README and additional installs: 

sudo apt-get install libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio

None of these seems to be working. Any other ideas maybe?

arnlen, 2021-11-22:
Same here. To move forward, here is the output of the `rpiplay -h` command:

```
RPiPlay 1.2: An open-source AirPlay mirroring server for Raspberry Pi
Usage: rpiplay [-n name] [-b (on|auto|off)] [-r (90|180|270)] [-l] [-a (hdmi|analog|off)] [-vr renderer] [-ar renderer]
Options:

[...]

-vr renderer          Set video renderer to use. Available renderers:
    dummy: Dummy renderer; does not actually display video [Default]
-ar renderer          Set audio renderer to use. Available renderers:
    dummy: Dummy renderer; does not actually play audio [Default]
```

The `dummy` renderer is the only available, which mean that - indeed - gstreamer isn't available.

❓ Where does RPiPlay look to find gstreamer? An env variable to set maybe?

FD-, 2021-11-22:
Please post the cmake output.

meaurieati, 2021-11-22:
I've got the same issue. cmake outputs the following:
cmake ..
-- OpenMAX libraries not found, skipping compilation of Raspberry Pi renderer
-- Checking for modules 'gstreamer-1.0>=1.4;gstreamer-sdp-1.0>=1.4;gstreamer-video-1.0>=1.4;gstreamer-app-1.0>=1.4'
--   Found gstreamer-1.0, version 1.18.4
--   Found gstreamer-sdp-1.0, version 1.18.4
--   Found gstreamer-video-1.0, version 1.18.4
--   Found gstreamer-app-1.0, version 1.18.4
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/RPiPlay/build

and make -j this:
make -j
Scanning dependencies of target renderers
[ 11%] Built target llhttp
[ 28%] Built target playfair
[ 34%] Building C object renderers/CMakeFiles/renderers.dir/video_renderer_dummy.c.o
[ 34%] Building C object renderers/CMakeFiles/renderers.dir/audio_renderer_dummy.c.o
[ 37%] Building C object renderers/CMakeFiles/renderers.dir/video_renderer_gstreamer.c.o
[ 40%] Building C object renderers/CMakeFiles/renderers.dir/audio_renderer_gstreamer.c.o
[ 91%] Built target airplay
/home/pi/RPiPlay/renderers/video_renderer_gstreamer.c:33:17: warning: ‘check_plugins’ defined but not used [-Wunused-function]
   33 | static gboolean check_plugins(void)
      |                 ^~~~~~~~~~~~~
/home/pi/RPiPlay/renderers/audio_renderer_gstreamer.c:34:17: warning: ‘check_plugins’ defined but not used [-Wunused-function]
   34 | static gboolean check_plugins(void)
      |                 ^~~~~~~~~~~~~
[ 94%] Linking C static library librenderers.a
[ 94%] Built target renderers
[ 97%] Building CXX object CMakeFiles/rpiplay.dir/rpiplay.cpp.o
[100%] Linking CXX executable rpiplay
[100%] Built target rpiplay

8bignic8, 2021-11-23:
Same Problem on Rpi Bullseye and Ubuntu Server 21.10, please help :). And thanks btw. for the nice Program :)

8bignic8, 2021-11-23:
ok maybe stupid but when I start it with sudo  I get: 
 XDG_RUNTIME_DIR not set in the environment.

does this help?

arnlen, 2021-11-23:
☝️ One remarque after reading the cmake output: I notice that gstreamer is check at the beginning of the compilation.
Since I didn't have gstreamer installed yesterday when I compiled rpiPlay, I should try another cmake now that it is installed.
I keep you posted.

giansegato, 2021-11-23:
Update. I installed GStreamer (my bad, I didn't realize it was a needed requirement). It definitely works, but streaming doesn't work yet. What I did:

1. Installed GStreamer

```
sudo apt-get install libx264-dev libjpeg-dev -y
sudo apt-get install libgstreamer1.0-dev      libgstreamer-plugins-base1.0-dev      libgstreamer-plugins-bad1.0-dev      gstreamer1.0-plugins-ugly      gstreamer1.0-tools      gstreamer1.0-gl      gstreamer1.0-gtk3 --fix-missing -y
```

2. Tested it

```
gst-launch-1.0 videotestsrc ! videoconvert ! autovideosink
```

I successfully saw a test stream on the target screen!

3. Re-compile RPiPlay

```
cmake --DCMAKE_CXX_FLAGS="-O3" --DCMAKE_C_FLAGS="-O3" ..
make -j
```

Results:

```
-- OpenMAX libraries not found, skipping compilation of Raspberry Pi renderer
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/RPiPlay/build
[ 28%] Built target llhttp
[ 28%] Built target playfair
[ 42%] Built target renderers
[ 94%] Built target airplay
[100%] Built target rpiplay
```

4. Launch it

```
rpiplay -n "TV" -a hdmi
```

Unfortunately, it still doesn't show anything, even if GStreamer does indeed work.

davidb0, 2021-11-23:
I had the same issue after updating my Pi Zero W to Raspberry Pi OS Bullseye yesterday. I reinstalled Buster (the previous version) and it works now. 

The issues I had was: 
the "cmake .." was way too fast and the renders RPI (Omax) and Gstreamer didn't work and didn't get compiled.
the iPhone connected to it, but nothing happened on screen, no sound no audio.
when I tried installing Gstreamer on Bullseye, it didn't work (maybe I'm a noob, but whatever).

FD-, 2021-11-23:
So this sounds like it is somewhat related to #294

giansegato, 2021-11-25:
If it can help, I downloaded Raspian Buster Lite (2020), and followed the entire procedure again. It worked!

abdullahaltun70, 2021-11-25:
> If it can help, I downloaded Raspian Buster Lite (2020), and followed the entire procedure again. It worked!

where did you download buster lite en what steps did you exactly follow? I would really like to try according to your taken steps.

8bignic8, 2021-11-25:
> > If it can help, I downloaded Raspian Buster Lite (2020), and followed the entire procedure again. It worked!
> 
> where did you download buster lite en what steps did you exactly follow? I would really like to try according to your taken steps.

Yes me 2 :)

abdullahaltun70, 2021-11-25:
I have installed Buster Lite. installed everything that was needed. Up untill the "make -j" command everything went smooth. While executing the command I get the following errors -> any Idea how to fix this??:
cc1plus: out of memory allocating 65536 bytes after a total of 720896 bytes

cc1plus: out of memory allocating 65536 bytes after a total of 688128 bytes

cc1plus: out of memory allocating 65536 bytes after a total of 745472 bytes

cc1plus: out of memory allocating 65536 bytes after a total of 741376 bytes
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
virtual memory exhausted: Cannot allocate memory
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1233: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSBRdec/src/pvc_dec.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
virtual memory exhausted: Cannot allocate memory
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:63: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libAACdec/src/FDK_delay.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:245: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libAACdec/src/conceal.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:349: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libAACdec/src/usacdec_ace_ltp.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:414: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libAACdec/src/usacdec_rom.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:453: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libDRCdec/src/drcDec_gainDecoder.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:557: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libFDK/src/FDK_bitbuffer.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:687: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libFDK/src/autocorr2nd.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:934: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSACdec/src/sac_bitdec.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1025: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSACdec/src/sac_reshapeBBEnv.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:960: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSACdec/src/sac_dec.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1051: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSACdec/src/sac_smoothing.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1077: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSACdec/src/sac_tsd.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1181: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSBRdec/src/psbitdec.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1220: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSBRdec/src/psdecrom_drm.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1272: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSBRdec/src/sbr_ram.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1298: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSBRdec/src/sbrdec_drc.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1246: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSBRdec/src/sbr_deb.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1324: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSBRdec/src/sbrdecoder.cpp.o] Error 1
make[2]: *** [renderers/fdk-aac/CMakeFiles/fdk-aac.dir/build.make:1350: renderers/fdk-aac/CMakeFiles/fdk-aac.dir/libSYS/src/syslib_channelMapDescr.cpp.o] Error 1

arnlen, 2021-11-25:
Out of memory issue... Which is your Raspi model?

---

_Quick update on my side_
- gstreamer installed
- `cmake ..` + `make -j` + `sudo make install` + `sudo reboot` and boom, new error:

```
rpiplay: error while loading shared libraries: libopenmaxil.so: cannot open shared object file: No such file or directory
```

I've also tried a `sudo rpi-update`, but still doesn't work.

abdullahaltun70, 2021-11-25:
> Out of memory issue... Which is your Raspi model?
>
>
>
> ---
>
>
>
> _Quick update on my side_
>
> - gstreamer installed
>
> - `cmake ..` + `make -j` + `sudo make install` + `sudo reboot` and boom, new error: [...]

Raspberry 2

meaurieati, 2021-11-25:
Installing Raspbian buster lite (2020) worked for me too!

giansegato, 2021-11-26:
> > If it can help, I downloaded Raspian Buster Lite (2020), and followed the entire procedure again. It worked!
> 
> where did you download buster lite en what steps did you exactly follow? I would really like to try according to your taken steps.

Downloaded it [here](https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/). Remember to do a full `apt update && apt upgrade` since there's gonna be a lot of packages that needs to be updated (especially security ones).

sternstef, 2021-11-26:
Out of memory issue... Which is your Raspi model?

sternstef, 2021-11-26:
Out of memory issue... Which is your Raspi model?
---

In case of memory issues during compilation, try `make -j1` or `make -j2`. The reduced parallelism worked for me.

abdullahaltun70, 2021-11-27:
> > Out of memory issue... Which is your Raspi model?
> 
> In case of memory issues during compilation, try `make -j1` or `make -j2`. The reduced parallelism worked for me.

It worked!! the steps I took were as follow:

1. download the Raspbian Buster .img from this [link](https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/). (I downloaded the .zip file).
2. I used Raspberry Pi manager to install Buster onto my SD card.
3. I needed to install Git as it wasn't preinstalled somehow. 
4. I executed the commands:
git clone https://github.com/FD-/RPiPlay.git
cd RPiPlay
sudo apt-get install cmake
sudo apt-get install libavahi-compat-libdnssd-dev
sudo apt-get install libplist-dev
sudo apt-get install libssl-dev
mkdir build
cd build
cmake ..
make -j

4.5. Instead of "make -j" I used "make -j2" because I encountered some memory allocating issues. 
5. I started the server with "./rpiplay -n Monitor -a hdmi"

It went to a black screen as said and I could finally mirror my IPhone.

Hope this helps

wdcossey, 2021-11-29:
Finally got it working with my `Pi Zero 2 W` using `Bullseye Lite`, I'll write what I did below step-by-step (feel free to chain the commands if you desire).

> Update your OS
```bash
apt update
apt upgrade
```

> Get `OpenMAX` from the firmware package, thanks @pallas 
```bash
rpi-update
```
> reboot your pi

> Install `GStreamer`, thanks @8bignic8
```bash
apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev
```

> Install git, pull this repo and cmake
```bash
apt-get install git
git clone https://github.com/FD-/RPiPlay.git
cd RPiPlay
sudo apt-get install cmake
sudo apt-get install libavahi-compat-libdnssd-dev
sudo apt-get install libplist-dev
sudo apt-get install libssl-dev
mkdir build
cd build
cmake --DCMAKE_CXX_FLAGS="-O3" --DCMAKE_C_FLAGS="-O3" ..
```

> Output of `cmake`
```text
-- Found OpenMAX libraries for Raspberry Pi
-- Checking for modules 'gstreamer-1.0>=1.4;gstreamer-sdp-1.0>=1.4;gstreamer-video-1.0>=1.4;gstreamer-app-1.0>=1.4'
--   Found gstreamer-1.0, version 1.18.4
--   Found gstreamer-sdp-1.0, version 1.18.4
--   Found gstreamer-video-1.0, version 1.18.4
--   Found gstreamer-app-1.0, version 1.18.4
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/RPiPlay/build
```

> run `make` (using `-j2` for the Zero [2])
```
make -j2
```

> Running `./rpiplay` now will most likely result in an error (I had this issue on the `Zero 2`)

```text
error while loading shared libraries libbrcmGLESv2 (or similar)
```

> Correcting the symbolic link (from https://raspberrypi.stackexchange.com/a/133496 by `DL1MX`)

```bash
apt install libgles-dev libegl-dev
sudo ln -s /usr/lib/arm-linux-gnueabihf/libGLESv2.so /usr/lib/libbrcmGLESv2.so
sudo ln -s /usr/lib/arm-linux-gnueabihf/libEGL.so /usr/lib/libbrcmEGL.so
```

> You should now be able to run `./rpiplay` i.e. `./rpiplay -n "AirPlay" -vr rpi -ar rpi -a hdmi` 
> Alas, you will most likely still not get any output from HDMI (just a black screen)

> Time to edit your `config.txt`
```
sudo nano /boot/config.txt
```

> Comment out `dtoverlay=vc4-kms-v3d`

> i.e
```text
# Enable DRM VC4 V3D driver
# dtoverlay=vc4-kms-v3d
```

> Optionally you can change `dtoverlay` to `vc4-fkms-v3d` (NOTE: `fkms` not `kms`)

> Save the file, reboot your pi
```bash
sudo reboot
```

> Navigate back to your buid directory `cd RPiPlay/build/` and run `./rpiplay` i.e. `./rpiplay -n "AirPlay" -vr rpi -ar rpi -a hdmi`

8bignic8, 2021-11-29:
> sudo nano /boot/config.txt

I tried that but I got: 

aacDecoder_DecodeFrame error : 0x4002

8bignic8, 2021-11-30:
If you only need Audio, that worked directly, https://github.com/juhovh/shairplay

adoconnection, 2021-12-10:
@FD-  consider updating readme please
It was confusing for me that PriPlay can mirror but cannot accept video stream like youtube.

For my RPi 3B+ I had to make following steps:
@giansegato comments combined into one

1. Burn this particular image of buster lite: 2020-02-13 (bullseye will not work, for now dec 2021)
```
https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip
```
2. update and upgrade
```
apt update
apt upgrade
```
3. install GStreamer
there is one more package I had to install: ```gstreamer1.0-plugins-good```
```
apt-get install \
     libx264-dev 
     libjpeg-dev
     libgstreamer1.0-dev
     libgstreamer-plugins-base1.0-dev
     libgstreamer-plugins-bad1.0-dev 
     gstreamer1.0-plugins-ugly
     gstreamer1.0-tools
     gstreamer1.0-gl
     gstreamer1.0-gtk3
     gstreamer1.0-plugins-good --fix-missing -y
```
```gst-launch-1.0 videotestsrc ! videoconvert ! autovideosink``` did nothing for me, it can be skipped

4. install RpiPlay
```
apt-get install git cmake libavahi-compat-libdnssd-dev libplist-dev libssl-dev -y
```
```
git clone https://github.com/FD-/RPiPlay.git
cd RPiPlay
mkdir build
cd build
cmake ..
make -j
```

5. run 
```
./rpiplay -n "SkyView" -a hdmi
```

FD-, 2021-12-10:
Just for clarity: If you use Buster on a Raspberry Pi, you won't need any of the gstreamer stuff.

adoconnection, 2021-12-10:
> Just for clarity: If you use Buster on a Raspberry Pi, you won't need any of the gstreamer stuff.

i will try again, but im almost sure it did not work at a time I tried.
if you enable wiki updates I can make a instruction page

brechtknecht, 2021-12-13:
[image]

This issue was resolved for me by selecting the default video output option in `sudo raspi-config`.

`2 Display Options → D1 Resolution → Default`

DaCapitalist, 2021-12-15:
> 

Hi, thanks for the tutorials here!
Could you please help to run RPiPlay on Docker in Raspberry PI OS 64bit Bullseye?
I tried to adjust a Dockerfile based on @jhfdevelop 's and @wdcossey 's contributions, but have issues with the libdns_sd.so.1 library:
https://github.com/DaCapitalist/rpiplay_docker/blob/main/Dockerfile
```
Step 6/21 : RUN mv ./firmware-master/opt/vc/ /opt
 ---> Running in 7fd595b6ff3b
Removing intermediate container 7fd595b6ff3b
 ---> b24783f451f8
Step 7/21 : RUN git clone https://github.com/FD-/RPiPlay.git
 ---> Running in 7c751c04890c
Cloning into 'RPiPlay'...
Removing intermediate container 7c751c04890c                                                                        
 ---> 3683a697717c
Step 8/21 : WORKDIR /work/RPiPlay
 ---> Running in dc16dd50a653
Removing intermediate container dc16dd50a653
 ---> 5ef1592e0102
Step 9/21 : RUN mkdir build
 ---> Running in 45c77a34927e
Removing intermediate container 45c77a34927e
 ---> c0b3e01869c3
Step 10/21 : RUN cp /usr/lib/arm-linux-gnueabihf/libdns_sd.so.1 ./build/
 ---> Running in eb7a361aced0
cp: cannot stat '/usr/lib/arm-linux-gnueabihf/libdns_sd.so.1': No such file or directory
The command '/bin/sh -c cp /usr/lib/arm-linux-gnueabihf/libdns_sd.so.1 ./build/' returned a non-zero code: 1
```

I see that this worked for you already. What am I doing wrong?

I see that a PR is already pending but it seems not working for me. Any chance to fix it and pull it so that we could make use of it?

Thanks!

DaCapitalist, 2021-12-15:
Hi, thanks for the tutorials here!
Could you please help to run RPiPlay on Docker in Raspberry PI OS 64bit Bullseye?
I tried to adjust a Dockerfile based on @jhfdevelop 's and @wdcossey 's contributions, but have issues with the libbrcmGLESv2.so library:
https://github.com/DaCapitalist/rpiplay_docker/test/Dockerfile
```
[100%] Linking CXX executable rpiplay
/usr/bin/ld: /opt/vc/lib/libbrcmGLESv2.so: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status                                                                          
make[2]: *** [CMakeFiles/rpiplay.dir/build.make:119: rpiplay] Error 1                                               
make[1]: *** [CMakeFiles/Makefile2:227: CMakeFiles/rpiplay.dir/all] Error 2                                         
make: *** [Makefile:149: all] Error 2                                                                               
The command '/bin/sh -c cd build && cmake --DCMAKE_CXX_FLAGS="-O3" --DCMAKE_C_FLAGS="-O3" .. && make' returned a non-zero code: 2
```

Is that due to the fact that you described in another issue regarding 64bit libraries not being available?
What do I need to change in the Dockerfile to either use 32bit libraries within the container instead or fallback to a gstreamer-only solution?

Thanks!

Ccode-lang, 2022-01-14:
@adoconnection Was right.  Installing that fixes all your problems.

albocc, 2022-01-18:
@FD- Could you please update the README so the guide works with the latest Raspberry Pi image? (bullseye presumably?)
When I saw the project I expected the steps to just work with the latest (lite) image. Now I am reading these issues and am told to use an old raspbian version instead, which is not a solution in my book.

Ccode-lang, 2022-01-27:
@albocc does Raspbian Lite come with a window manager?  If not that may be your problem.

albocc, 2022-01-27:
@Ccode-lang It does not, that is what the lite version is for. But I do not think that is necessary because I tried a docker image somebody else built which seemed to work. However, I would like to build it myself. 
Also, some people in this issue reported that they could get it to work with the lite edition, so it should be possible. 🤔

bad2uXX, 2022-01-30:
> It worked!! the steps I took were as follow:
>
> 1. download the Raspbian Buster .img from this [link](https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/). (I downloaded [...]

I have same issue with Raspberry Pi B+ with latest Raspberry Pi OS Lite ( release 28 January  2022), so I've decided to try with Buster and it worked. Thanks for detailed steps.

albocc, 2022-02-08:
@bad2uXX "so I've decided to try with Buster and it worked." That's great for now, but the goal has to be to fix the guide for the latest release of Raspbian. At one point Buster will no longer be old-stable but instead deprecated and then this project stops working? Simply pointing to the old distro release is not a solution @FD-

closed 2022-02-25 by pallas

JSAnyone, 2022-08-13:
I also don't get it to work.
I've definitely installed streamer, and `gst-launch-1.0 videotestsrc ! videoconvert ! autovideosink` results in a test picture being displayed on the cdmi output.
Also when compiling there are no errors:
```
cmake ..
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libcrypto.so (found suitable version "1.1.1n", minimum required is "1.1.1")  
-- OpenMAX libraries not found, skipping compilation of Raspberry Pi renderer
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for modules 'gstreamer-1.0>=1.4;gstreamer-sdp-1.0>=1.4;gstreamer-video-1.0>=1.4;gstreamer-app-1.0>=1.4'
--   Found gstreamer-1.0, version 1.18.4
--   Found gstreamer-sdp-1.0, version 1.18.4
--   Found gstreamer-video-1.0, version 1.18.4
--   Found gstreamer-app-1.0, version 1.18.4
-- Configuring done
-- Generating done
-- Build files have been written to: /root/RPiPlay/build
```
```
make -j
Scanning dependencies of target playfair
Scanning dependencies of target renderers
Scanning dependencies of target llhttp
[  2%] Building C object lib/playfair/CMakeFiles/playfair.dir/playfair.c.o
[  5%] Building C object lib/playfair/CMakeFiles/playfair.dir/hand_garble.c.o
[  8%] Building C object lib/playfair/CMakeFiles/playfair.dir/omg_hax.c.o
[ 11%] Building C object lib/playfair/CMakeFiles/playfair.dir/modified_md5.c.o
[ 14%] Building C object lib/playfair/CMakeFiles/playfair.dir/sap_hash.c.o
[ 17%] Building C object lib/llhttp/CMakeFiles/llhttp.dir/api.c.o
[ 20%] Building C object lib/llhttp/CMakeFiles/llhttp.dir/llhttp.c.o
[ 22%] Building C object lib/llhttp/CMakeFiles/llhttp.dir/http.c.o
[ 25%] Building C object renderers/CMakeFiles/renderers.dir/audio_renderer_dummy.c.o
[ 28%] Building C object renderers/CMakeFiles/renderers.dir/video_renderer_gstreamer.c.o
[ 31%] Building C object renderers/CMakeFiles/renderers.dir/audio_renderer_gstreamer.c.o
[ 34%] Building C object renderers/CMakeFiles/renderers.dir/video_renderer_dummy.c.o
/root/RPiPlay/renderers/video_renderer_gstreamer.c:33:17: warning: ‘check_plugins’ defined but not used [-Wunused-function]
   33 | static gboolean check_plugins(void)
      |                 ^~~~~~~~~~~~~
/root/RPiPlay/renderers/audio_renderer_gstreamer.c:34:17: warning: ‘check_plugins’ defined but not used [-Wunused-function]
   34 | static gboolean check_plugins(void)
      |                 ^~~~~~~~~~~~~
[ 37%] Linking C static library librenderers.a
[ 37%] Built target renderers
[ 40%] Linking C static library libplayfair.a
[ 40%] Built target playfair
[ 42%] Linking C static library libllhttp.a
[ 42%] Built target llhttp
Scanning dependencies of target airplay
[ 45%] Building C object lib/CMakeFiles/airplay.dir/byteutils.c.o
[ 48%] Building C object lib/CMakeFiles/airplay.dir/crypto.c.o
[ 51%] Building C object lib/CMakeFiles/airplay.dir/fairplay_playfair.c.o
[ 54%] Building C object lib/CMakeFiles/airplay.dir/dnssd.c.o
[ 57%] Building C object lib/CMakeFiles/airplay.dir/http_response.c.o
[ 60%] Building C object lib/CMakeFiles/airplay.dir/http_request.c.o
[ 62%] Building C object lib/CMakeFiles/airplay.dir/logger.c.o
[ 65%] Building C object lib/CMakeFiles/airplay.dir/httpd.c.o
[ 68%] Building C object lib/CMakeFiles/airplay.dir/mirror_buffer.c.o
[ 71%] Building C object lib/CMakeFiles/airplay.dir/netutils.c.o
[ 74%] Building C object lib/CMakeFiles/airplay.dir/pairing.c.o
[ 77%] Building C object lib/CMakeFiles/airplay.dir/raop_ntp.c.o
[ 80%] Building C object lib/CMakeFiles/airplay.dir/raop.c.o
[ 82%] Building C object lib/CMakeFiles/airplay.dir/raop_buffer.c.o
[ 85%] Building C object lib/CMakeFiles/airplay.dir/raop_rtp.c.o
[ 88%] Building C object lib/CMakeFiles/airplay.dir/utils.c.o
[ 91%] Building C object lib/CMakeFiles/airplay.dir/raop_rtp_mirror.c.o
[ 94%] Linking C static library libairplay.a
[ 94%] Built target airplay
Scanning dependencies of target rpiplay
[ 97%] Building CXX object CMakeFiles/rpiplay.dir/rpiplay.cpp.o
[100%] Linking CXX executable rpiplay
[100%] Built target rpiplay

```
And when executing RPIPlay there are no errors, but nothing the hdmi output continues showing the login prompt instead of turning black:
```
rpiplay -b on -vr gstreamer -ar gstreamer
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
```

Any ideas?

JSAnyone, 2022-08-13:
I've just noticed that the `cmake..` command is looking for `gstreamer-1.0>=1.4;gstreamer-sdp-1.0>=1.4;gstreamer-video-1.0>=1.4;gstreamer-app-1.0>=1.4`, but my gstreamer version is `1.18.4`.
Is there any way to make RPIPlay compatible for gstreamer `1.18.4`? It doesn't look like there is a newer gstreamer available for bullseye.

holubyev, 2022-10-27:
There no need to burn particular image and of course there is no need to have window manager hence the guide is about Raspbian Lite.
I managed to get working on my old RPi Br2 by downloading official Lite version via PI imager. This is my configuration>
```
Raspberry Pi Model B Rev 2
Raspbian GNU/Linux 11 (bullseye Lite)
Linux pi 5.15.74+ #1595 Wed Oct 26 11:01:05 BST 2022 armv6l GNU/Linux
```
I followed the @wdcossey's guide (thanks for that), with only one addition (and I think with only one exclusion `sudo apt install libgles-dev libegl-dev` was already installed).

The main problem with new Raspbian is, that there isn't firmware files that RPiPlay needs for compilation: `https://github.com/raspberrypi/firmware/tree/master/opt/vc/src/hello_pi/libs/ilclient`. `sudo rpi-update` will download the whole firmware including these files, but the linker didn't find them in the final phase of compilation (although ilclient was built successfully in early stage of `make -j2` command).
My solution was to add these files to ~/RPiPlay/renderers/ directory before `make -j2`:
```
cp /opt/vc/src/hello_pi/libs/ilclient/* ~/RPiPlay/renderers/
```
Hope it helps.

drk-rose, 2023-08-15:
> @FD- consider updating readme please It was confusing for me that PriPlay can mirror screen but cannot accept video stream like youtube.
> 
> For my RPi 3B+ and iphone 12 I had to make following steps over SSH: @giansegato comments combined into one
> 
> 1. Burn this particular image of buster lite: 2020-02-13 (bullseye will not work, for now dec 2021)
> 
> ```
> https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2020-02-14/2020-02-13-raspbian-buster-lite.zip
> ```
> 
>  2. update and upgrade
> 
> ```
> apt update
> apt upgrade
> ```
> 
>  3. install GStreamer
>    there is one more package I had to install: `gstreamer1.0-plugins-good`
> 
> ```
> apt-get install \
>      libx264-dev 
>      libjpeg-dev
>      libgstreamer1.0-dev
>      libgstreamer-plugins-base1.0-dev
>      libgstreamer-plugins-bad1.0-dev 
>      gstreamer1.0-plugins-ugly
>      gstreamer1.0-tools
>      gstreamer1.0-gl
>      gstreamer1.0-gtk3
>      gstreamer1.0-plugins-good --fix-missing -y
> ```
> 
> `gst-launch-1.0 videotestsrc ! videoconvert ! autovideosink` did nothing for me, it can be skipped
> 
> 4. install RpiPlay
> 
> ```
> apt-get install git cmake libavahi-compat-libdnssd-dev libplist-dev libssl-dev -y
> ```
> 
>  ```
> git clone https://github.com/FD-/RPiPlay.git
> cd RPiPlay
> mkdir build
> cd build
> cmake ..
> make -j
> ```
> 
>  5. run
> 
> ```
> ./rpiplay -n "SkyView" -a hdmi
> ```

### Coming to you from the year 2023!

I got this process to work on the Pi Zero 2 W, but had to resort to the [2021-05-28 version of Buster](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/).

At the top of the process on a fresh install, though, we of course had to get git:
`sudo apt install git`

And to put it altogether in the end, I opted for:
`make -j2`
`sudo make install`

To get it running and keep it running, though, it seems to only intermittently work unless low latency mode is on and the audio option for hdmi also has to be set:
`./rpiplay -a hdmi -l`

If you plan to have this launch on boot up, however, the same process from the RPiPlay wiki works, but you'll want to add those mentioned options right after the RPiPlay binary path.
`pi@raspberrypi:~$ nano .bashrc`

`# within an SSH-session, ${SSH_TTY} will be set`
`if [ "x${SSH_TTY}" = "x" ]; then`
`  /home/pi/rpiplay # Or wherever the rpiplay binary is located`
`fi`

Hope this helps to keep this project running for us all!

Why buy a Google Chromecast when you could spend 15$ and several hours to get the results you want and slide a comment on a GitHub chat board? Lol

a7mad3bdullah, 2023-09-27:
The only way I got it to work is by installing it on Raspberry Pi OS Buster not Bullseye. See [364](https://github.com/FD-/RPiPlay/issues/364)