FD-/RPiPlay #308: Using RPiPlay on Raspbian Bullseye

konrul-r10, 2021-12-19:
Hello,

I have tried to install & use RPiPlay on Raspbian Bullseye. Install part was an unsuccessful attempt therefore I brought files from the working buster (I got RPİPlay files from my old build, Raspbian Buster, to the Bullseye),  however, I couldn't run it. 

Briefly, is there any way to run this on bullseye? If so could you please help?

Thanks in advance. Best regards.

pallas, 2021-12-19:
See #304 #296 #294, which all already answer your question. tl;dr: It does work but you need to install OpenMAX again via rpi-update, which is a dangerous command (read: do so at your own risk). Then you need to clean out the build directory if you already tried to cmake the project, and then you need to fix a symlink for GLES.

closed 2021-12-19 by pallas

konrul-r10, 2022-01-11:
> See #304 #296 #294, which all already answer your question. tl;dr: It does work but you need to install OpenMAX again via rpi-update, which is a dangerous command (read: do so at your own risk). [...]

Hello pallas, Thanks for your help. I have tried to find a solution, including your suggestions but I am not successful. Is there any way to explain step by step how to do that? I would like to take the risk (actually I have already updated firmware.) As a beginner Linux user, step by step explanation will be helpful for me and other users I think.

pallas, 2022-01-12:
You ran https://github.com/raspberrypi/rpi-update and it didn't install the VC libs?

pallas, 2022-01-12:
You can also try to pull in https://github.com/raspberrypi/firmware/tree/master/opt/vc

konrul-r10, 2022-01-13:
> You ran https://github.com/raspberrypi/rpi-update and it didn't install the VC libs?

it seems that its already installed. but i am still having black screen issue and no audio...

a7mad3bdullah, 2023-09-27:
The only way I was able to make it work is by installing it on Raspberry Pi OS Buster Not Bullseye, see [364](https://github.com/FD-/RPiPlay/issues/364)

> > You ran https://github.com/raspberrypi/rpi-update and it didn't install the VC libs?
> 
> it seems that its already installed. but i am still having black screen issue and no audio...
> 
> I have tried to build one more time and made `sudo make install` tried to use like this: `#rpiplay -n Televizyon -a hdmi -ar rpi -l -b off -vr rpi`
> 
> I got this message: `rpiplay: error while loading shared libraries: libopenmaxil.so: cannot open shared object file: No such file or directory`
> 
> I have tried to use `sudo ln -s /opt/vc/libopenmaxil.so /usr/lib/libopenmaxil.so`
> 
> but no luck. Is there anything I missed?

The only way I was able to make it work is by installing it on Raspberry Pi OS Buster Not Bullseye, see [364](https://github.com/FD-/RPiPlay/issues/364)