FD-/RPiPlay #310: Rpiplay not showing the mirroring

Ency79, 2021-12-25:
Hello

I have an RPI3B wit ubuntu 20.04.03 and wanted to setup RPiPlay as documented in GitHub https://github.com/FD-/RPiPlay

I did not receive any errors during the installation and my RPI3 is connected to the same network like my iPhone or MacBook. But when executing on the rpi:
`rpiplay -n "AirPlay" -a hdmi -l -d `

My TV where the rpi3 is connected over hdmi is not going into a black screen when I try to mirror the iPhone or MacBook. On my iPhone screen I only see it’s connected to airplay but the RPI3 on the tv still shows the commandline and not a black screen…

in the command above I have added the option -d for debugging and that is the output:
`ubuntu@rpi3:~$ rpiplay -n "Neno's AirPlay" -a hdmi -l -d
Using low-latency mode
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 17
Local: 192.168.178.28
Remote: 192.168.178.35
httpd receiving on socket 17
conn_request
Handling request GET with URL /info
INFO len = -1417686036
httpd receiving on socket 17
conn_request
Handling request POST with URL /pair-setup
httpd receiving on socket 17
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 17
conn_request
Handling request POST with URL /pair-verify
httpd receiving on socket 17
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 17
conn_request
Handling request POST with URL /fp-setup
httpd receiving on socket 17
conn_request
Handling request SETUP with URL rtsp://192.168.178.28/15677326586792462425
DACP-ID: 8EBBDA3405CBE37A
Active-Remote: 890036244
Transport: null
SETUP 1
eiv_len = 16
ekey_len = 72
fairplay_decrypt ret = 0
timing_rport = 54213
raop_ntp parse remote ip = 192.168.178.35
raop_ntp starting time
raop_rtp parse remote ip = 192.168.178.35
raop_rtp_mirror parse remote ip = 192.168.178.35
eport = 45091, tport = 46786
raop_ntp send_len = 32
raop_ntp receive time type_t packetlen = 32
raop_ntp sync correction = -1640077992268392
Accepted IPv4 client on socket 22
Local: 192.168.178.28
Remote: 192.168.178.35
httpd receiving on socket 17
conn_request
Handling request GET with URL /info
INFO len = -1417686036
httpd receiving on socket 17
conn_request
Handling request GET_PARAMETER with URL rtsp://192.168.178.28/15677326586792462425
httpd receiving on socket 17
conn_request
Handling request RECORD with URL rtsp://192.168.178.28/15677326586792462425
raop_handler_record
httpd receiving on socket 17
conn_request
Handling request SET_PARAMETER with URL rtsp://192.168.178.28/15677326586792462425
httpd receiving on socket 17
conn_request
Handling request SETUP with URL rtsp://192.168.178.28/15677326586792462425
DACP-ID: 8EBBDA3405CBE37A
Active-Remote: 890036244
Transport: null
type = 110
streamConnectionID = 1692323531819992909
raop_rtp_mirror starting mirroring
Mirroring initialized successfully
raop_rtp_mirror accepting client
raop_rtp_mirror width_source = 450.000000 height_source = 972.000000 width = 470.000000 height = 1016.000000
raop_rtp_mirror sps size = 17
raop_rtp_mirror pps size = 4
raop_rtp_mirror video ntp = 1640450309426616, now = 1640450309359393, latency = -67223
raop_rtp_mirror video ntp = 1640450309476694, now = 1640450309390403, latency = -86291
raop_rtp_mirror video ntp = 1640450309493386, now = 1640450309397125, latency = -96261
raop_rtp_mirror video ntp = 1640450309510079, now = 1640450309412870, latency = -97209
raop_rtp_mirror video ntp = 1640450309526772, now = 1640450309429261, latency = -97511
raop_rtp_mirror video ntp = 1640450309543464, now = 1640450309445119, latency = -98345
raop_rtp_mirror video ntp = 1640450309560157, now = 1640450309465302, latency = -94855
raop_rtp_mirror video ntp = 1640450309576850, now = 1640450309514965, latency = -61885
raop_rtp_mirror video ntp = 1640450309593542, now = 1640450309521490, latency = -72052
raop_rtp_mirror video ntp = 1640450309610235, now = 1640450
Stopping...
raop_rtp_mirror video ntp = 1640450317355194, now = 1640450317255945, latency = -99249
raop_rtp_mirror video ntp = 1640450317371886, now = 1640450317313071, latency = -58815
Removing connection for socket 17
Destroying connection
raop_ntp stopping time thread
raop_ntp exiting thread
raop_ntp stopped time thread
raop_rtp_mirror video ntp = 1640450317388579, now = 1640450317334168, latency = -54411
raop_rtp_mirror exiting TCP thread
Removing connection for socket 22
Destroying connection
Exiting HTTP thread  `

for Me it looks like everything is ok, am I right??? But why I do not see my iPhone on the TV??? I also checked on the rpi the tty session and switched to all of them but still I see the commandline and the black screen will not appear.

many ideas what can be the problem here??

JYKeith123, 2021-12-26:
me neither. I'll try this #296. OS Bullseye -> Buster

JYKeith123, 2021-12-26:
It works! :-)

Ency79, 2021-12-26:
I think it’s not working cause of the ubuntu Server image which unused to install on the raspberry pi3b?!?

how I mentioned the setup of rpiplay worked without problems but when mirroring my iphone I don’t see the black screen on tv and also not that my iphone is mirroring

Ency79, 2021-12-27:
With the buster Image I was able to install rpiplay and also mirroring works. I also hear audio but quality very terrible. Is there any fix to optimize it??? Any driver or addon which needs to be installed???

claulsc, 2022-01-04:
Is it possible to share iPhone screen on TV without connecting the pi to the TV via HDMI? Pi its in another room and cant be psihically connected.

Ency79, 2022-01-04:
My RPi is connected via HDMI. I would also appreciate to have it without connecting the hdmi directly to the rpi. I know there are hdmi transmitters available but I dont know if that works on rpi. That would be interesting to know

pallas, 2022-01-04:
There isn't enough actionable information in this issue to do anything. It sounds like you just have your Pi misconfigured at the system level.

closed 2022-01-04 by pallas