FD-/RPiPlay #314: History/authorship: dsafa22 -> DroidFang?

fduncanh, 2022-01-08:
Looking through the history of the mirror code,  as documented in the interesting write-up https://www.itread01.com/hkeyxif.html, (translates well with Google Chrome) apparently guided by analysis of [this](https://aapks.com/apk/happycast/version/35810679/dl/)   and https://github.com/KqSMea8/AirplayServer , it look like the true author of the AirPlay2 mirror code (based on the audio code in shairplay) was "DroidFang", and dsafa22 merely created the github repository for DroidFang   (unless they are one and the same)

fduncanh, 2022-01-16:
Looks like dsafa22 and droidfang are the same person. (?)

 dsafa22 kept on developing AirplayServer  after the fork to RPiPlay.

latest I saw is dsafa22  commit history up to 2019-08-28 in preserved at https://github.com/jiangban/AirplayServer

closed 2022-02-01 by fduncanh

FD-, 2022-02-01:
Yes, back when I started RPiPlay, dsafa22 was still actively working on their branch. Then one day the original dsafa22/AirplayServer including all recent commits disappeared and was replaced by the current one.

fduncanh, 2023-03-31:
A while back I fixed up the translation of what I think is the mirror code author's  (dsafa22 / droid fang?)  article about how he did it.

Its on the UxPlay Wiki https://github.com/FDH2/UxPlay/wiki/AirPlay2

I also found  the Android app he analyzed : com.hpplay.happyplay 7.1.0 APK