FD-/RPiPlay #319: "Unable to connect to RPiPlay"

Man-with-Arrow, 2022-01-15:
RPiPlay compiles fine, runs fine, but does not detect an attempted connection. Output of `./rpiplay` and `./rpiplay -d` is identical:

```
Initialized server socket(s)

*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.

*** WARNING *** Please fix your application to use the native API of Avahi!

*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
```

RPiPlay shows up on iPad (running the latest OS 15.2), but connection silently times out. No indication of anything on PC, notification on iPad shows "Unable to connect to RPiPlay"

Running fully-updated Fedora 35, no firewall. 

Running RPiPlay on a laptop connected via WiFi, also on Fedora 35, works perfectly fine.

Is this a known issue? Any pointers on further debugging?

fduncanh, 2022-01-17:
next thing you should see is
```
Accepted IPv4 client on socket XX
```

Since your dont see it, rpiplay has not heard your iOS client's connection request.

This is either because of a firewall (are you *really* sure there isnt one active?) or a network problem.
Either way, its not an rpiplay problem.

Man-with-Arrow, 2022-01-17:
Unless the default Fedora install has a firewall, I don't have one... Sorry for the novice question, but how could I verify?

closed 2022-01-17 by Man-with-Arrow

fduncanh, 2022-01-17:
@Man-with-Arrow 
If you want to keep using a firewall, see htttp://github.com/FDH2/UxPlay    (RPiPlay derivative  aimed at desktop computers, but should be OK with a RPi 4.)

Man-with-Arrow, 2022-01-18:
That's what I ended up using :-)

Major props on maintaining both projects, by the way. I was surprised at how seamlessly the work.