FD-/RPiPlay #320: Video mirroring not working

mariomdra, 2022-01-23:
After installing Rpiplay on my rpi3B, everythings seems to work fine, and my macbook connects to the device. However, the black screen does not appear on the monitor when I click enter, so the macbook screen is not mirrored to the rpiplay.

catalinescu-sudo, 2022-01-25:
I have same issue with iphone and imac. This is my second raspberry pi 4, on the first one is working and the install steps was the same, in the same envoirment. Maybe is an issue from latest updates?
I get this error when installing at make -j
[ 34%] Building C object renderers/CMakeFiles/renderers.dir/video_renderer_gstreamer.c.o
/home/pi/RPiPlay/renderers/video_renderer_gstreamer.c:33:17: warning: ‘check_plugins’ defined but not used [-Wunused-function]
   33 | static gboolean check_plugins(void)
      |                 ^~~~~~~~~~~~~
/home/pi/RPiPlay/renderers/audio_renderer_gstreamer.c:34:17: warning: ‘check_plugins’ defined but not used [-Wunused-function]
   34 | static gboolean check_plugins(void)
      |                 ^~~~~~~~~~~~~
[ 37%] Linking C static library librenderers.a

Any help? Thank you!

fduncanh, 2022-01-28:
@catalinescu-sudo 
these are warnings not errors. not important.

If you want to use gstrreamer renderers on rpi4, consider testing [UxPlay](http://github.com/FDH2/UxPlay) and reporting there whether performance is OK on an RPi4 (this an enhanced gstreamer-based RPiPlay variant aimed at Desktop Linux).

closed 2022-02-25 by pallas