FD-/RPiPlay #321: rpiplay: error while loading shared libraries: libopenmaxil.so: cannot open shared object file: No such file or directory

JisatsuLove, 2022-01-29:
Hello,

I have a Problem with running RPiPlay.

I installed RPiPlay on a freshly installed Raspbian Bullseye Lite Pi 4 kernel: Linux raspberrypi 5.10.94-v7l+.

Followed the Instructions from the Read-Me.
When I try to start the RPiPlay with `rpiplay -n AppleTV -b off -a hdmi -vr rpi -ar rpi -d"`

I get the error code:
`rpiplay: error while loading shared libraries: libopenmaxil.so: cannot open shared object file: No such file or directory`

A quick look into `/opt/vc/lib` showed me
```
drwxr-xr-x 4 root root   4096 Jan 29 19:36 .
drwxr-xr-x 7 root root   4096 Jan 29 19:36 ..
-rw-r--r-- 1 root root  98612 Jan 29 19:36 libbcm_host.so
-rw-r--r-- 1 root root 202072 Jan 29 19:36 libbrcmEGL.so
-rw-r--r-- 1 root root 105768 Jan 29 19:36 libbrcmGLESv2.so
-rw-r--r-- 1 root root  99208 Jan 29 19:36 libbrcmOpenVG.so
-rw-r--r-- 1 root root  78552 Jan 29 19:36 libbrcmWFC.so
-rw-r--r-- 1 root root 364540 Jan 29 19:36 libcontainers.so
-rw-r--r-- 1 root root  16920 Jan 29 19:36 libdebug_sym.so
-rw-r--r-- 1 root root  14218 Jan 29 19:36 libdebug_sym_static.a
-rw-r--r-- 1 root root  66684 Jan 29 19:36 libdtovl.so
-rw-r--r-- 1 root root 202072 Jan 29 19:36 libEGL.so
-rw-r--r-- 1 root root 245356 Jan 29 19:36 libEGL_static.a
-rw-r--r-- 1 root root 344732 Jan 29 19:36 libelftoolchain.so
lrwxrwxrwx 1 root root     12 Jan 29 19:36 libGLESv1_CM.so -> libGLESv2.so
-rw-r--r-- 1 root root 105768 Jan 29 19:36 libGLESv2.so
-rw-r--r-- 1 root root 100076 Jan 29 19:36 libGLESv2_static.a
-rw-r--r-- 1 root root  13548 Jan 29 19:36 libkhrn_client.a
-rw-r--r-- 1 root root 450544 Jan 29 19:36 libkhrn_static.a
-rw-r--r-- 1 root root  68956 Jan 29 19:36 libmmal_components.so
-rw-r--r-- 1 root root  82388 Jan 29 19:36 libmmal_core.so
-rw-r--r-- 1 root root  15368 Jan 29 19:36 libmmal.so
-rw-r--r-- 1 root root  91696 Jan 29 19:36 libmmal_util.so
-rw-r--r-- 1 root root  44644 Jan 29 19:36 libmmal_vc_client.so
-rw-r--r-- 1 root root  33208 Jan 29 19:36 libopenmaxil.so
-rw-r--r-- 1 root root  99200 Jan 29 19:36 libOpenVG.so
-rw-r--r-- 1 root root  30288 Jan 29 19:36 libvchiq_arm.so
-rw-r--r-- 1 root root 108788 Jan 29 19:36 libvchostif.a
-rw-r--r-- 1 root root  28462 Jan 29 19:36 libvcilcs.a
-rw-r--r-- 1 root root 169364 Jan 29 19:36 libvcos.so
-rw-r--r-- 1 root root  31672 Jan 29 19:36 libvcsm.so
-rw-r--r-- 1 root root  78552 Jan 29 19:36 libWFC.so
drwxr-xr-x 2 root root   4096 Jan 29 19:36 pkgconfig
drwxr-xr-x 2 root root   4096 Jan 29 19:36 plugins
```

As far as i see the libary file `libopenmaxil.so` seems to be correctly installed.

Can someone tell me which libary path rpiplay tries to get if its trying to start?
Or can someone can tell me how to fix this easily ? :)

Any help is appreciated and thank you in regards

chriscasola, 2022-01-30:
I have also been unable to get this project to work with Raspbian Bullseye.

I think since openmax was removed in bullseye, the only audio sink supported by this project would be gstreamer, but I have been unable to get that to work either.

JisatsuLove, 2022-01-31:
I now went with the easier solution, downgrade the OS to Raspbian OS Buster so the files which are needed are preinstalled and now RPiPlay just works fine. I dont have problems with like Homeassistant or anything else too so i am fine with that Solution.

pallas, 2022-02-25:
openmaxil does not work on rpi 4, on older pi's you have to use rpi-update to reinstall the libraries but note that this is a dangerous operation; see all the other opened issues that already address this

closed 2022-02-25 by pallas