FD-/RPiPlay #332: Can't connect from MacBook Pro 13-inch, 2016 macOS Monterey 12.2.1 (21D62)

hipsterweeds, 2022-03-04:
[image]

fduncanh, 2022-03-12:
This could be a modified AirPlay protocol in Monterrey....

but there is a report https://github.com/FDH2/UxPlay/issues/67 that says a Monterrey mac worked with UxPlay (which is based on the code used by RPiPlay).    Please see if [UxPlay](https://github/com/FDH2/UxPlay) works for you, and report you findings there.

FD-, 2022-03-12:
You appear to be trying to connect to RPiPlay through the audio settings. This doesn't work, since RPiPlay is an AirPlay mirroring server (different protocols). Connect through the display settings in macOS.

fduncanh, 2022-03-13:
@FD- 
There is  a similar report on UxPlay

https://github.com/FDH2/UxPlay/issues/73

Could be that Monterrey no longer will use legacy protocol: nothing seem to happens after FPsetup.
(Uxplay's -d option reports all the conversation between client and server during  raop startup)
This would be bad news for the future...

fduncanh, 2022-04-01:
@hipsterweeds 

I confirmed that M1 macs work fine with RPiPlay.

The problem in UxPlay (now fixed) where the connection was made, but GStreamer failed to open a video window
came from a change made in UxPlay, and was not present in RPiPlay
(it was related to from a change made by dsafa22 after his code was forked to create RPiPlay, that had been ported to UxPlay).

Perhaps @hipsterweeds 's problem was something else.