FD-/RPiPlay #333: Got NULL buffer!

adminy, 2022-03-10:
I'm trying to use pi as a second screen for my mac.

I've placed my monitor vertical but for some reason that is not using the vertical screen, its trying to fit into a horizontal resolution.

Anyways, the mouse is super laggy even on small resolutions and it crashes with "Got NULL buffer!" within the first few seconds of being connected. No other errors even with debug mode on.

gregd72002, 2022-10-24:
Same problem here

pallas, 2022-10-24:
You’re out of memory. Please make sure your graphics split is large enough.

gregd72002, 2022-10-24:
@pallas, I've increased it from 64 to 256 and still getting the NULL buffer error. I should have mentioned that the problem was not there when the RPi was on Wifi. Now it is connected to via ethernet.