FD-/RPiPlay #334: Video4Linux on RPi working now on UxPlay: could be backported to RPiPlay.

fduncanh, 2022-03-12:
GStreamer OpenMAX plugin omxh264dec was broken for a long time, but now the development branch of GStreamer has got
Video4Linux plugin v4l2h264dec  working for hardware accelerated video on  the R  Pi GPU.   This is the one to be supported in the future on Pi.

see https://github.com/FDH2/UxPlay/issues/70    The GStreamer-based RPiPlay offspring [UxPlay](https://github.com/FDH2/UxPlay) on  RPi OS bullseye (32 or 64 bit) and Manjaro-Arm-RPi4 works with the patched plugin.  (backports to GStreamer 1.18.4 (bullseye) and 1.20.0 (Manjaro) are available)

FD-, 2022-03-12:
Very interesting indeed!

fduncanh, 2022-04-25:
Here are some [reports of success]( https://github.com/FDH2/UxPlay/wiki/UxPlay-on-Raspberry-Pi:-success-reports:) of UxPlay with Raspberry Pi model 4B using Video4Linux2 for  GPU h264 decoding

kang000feng, 2022-06-26:
> Very interesting indeed!

Dear developer,good day! May I ask how to Make sure the DUMP flags are not active ? How to check it? Thank you

fduncanh, 2022-06-26:
No idea what DUMP flags are.  You can test UxPlay on RPi with Video4Linux2 at the UxPlay site 
https://github.com/FDH2/UxPlay . you will need to patch the gstreamer Video4Linux2 plugin as explained there, until GStreamer-1.22 is released.

kang000feng, 2022-09-06:
> No idea what DUMP flags are. You can test UxPlay on RPi with Video4Linux2 at the UxPlay site https://github.com/FDH2/UxPlay . you will need to patch the gstreamer Video4Linux2 plugin as explained there, until GStreamer-1.22 is released. This is for RPi OS Bullseye, Ubuntu 22.04, etc. There is no support for older GStreamer < 1.18.4, as in a Debian Buster-based OS with 1.16.2

Dear developer, DUMP flags is from your README.md,  quote"For best performance:

Use a wired network connection
Compile with -O3 (cmake --DCMAKE_CXX_FLAGS="-O3" --DCMAKE_C_FLAGS="-O3" ..)
Make sure the DUMP flags are not active
Make sure you don't use the -d debug log flag
Make sure no other demanding tasks are running (this is particularly important for audio on the Pi Zero)"

fduncanh, 2022-09-06:
@kang000feng 
RpiPlay and [UxPlay](http://github.com/FDH2/UxPlay) are now different projects.  UxPlay supports  RPi using gstreamer and video4linux2.(32 and 64 bit)    RpiPlay supports RPi using Omx (32 bit) / If you need  UxPlay support, and its documentation doesnt solve it, post your issue there.