FD-/RPiPlay #347: Raspberry Pi 3 getting errors during make -j

JoeLajoe, 2022-09-04:
Getting Errors during make -j

Output:
 make -j1 PLATFORM=rpi1
[100%] Built target h264-bitstream
[100%] Built target playfair
[100%] Built target llhttp
[100%] Built target airplay
[100%] Built target ilclient
[100%] Built target fdk-aac
[100%] Built target renderers
[100%] Linking CXX executable rpiplay
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audi                                                                                                                                                             o_renderer_rpi_start':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:280: undefined reference to `ilc                                                                                                                                                             lient_change_component_state'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:281: undefined refe                                                                                                                                                             rence to `ilclient_enable_port_buffers'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:282: undefined refe                                                                                                                                                             rence to `ilclient_change_component_state'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:277: undefined refe                                                                                                                                                             rence to `ilclient_change_component_state'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audi                                                                                                                                                             o_renderer_rpi_set_volume':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:371: undefined reference to `ilc                                                                                                                                                             lient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:371: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audi                                                                                                                                                             o_renderer_rpi_render_buffer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:349: undefined reference to `ilc                                                                                                                                                             lient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:349: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:331: undefined refe                                                                                                                                                             rence to `ilclient_get_input_buffer'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audi                                                                                                                                                             o_renderer_rpi_destroy_renderer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:98: undefined reference to `ilcl                                                                                                                                                             ient_disable_tunnel'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:99: undefined refer                                                                                                                                                             ence to `ilclient_disable_port_buffers'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:100: undefined refe                                                                                                                                                             rence to `ilclient_teardown_tunnels'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:102: undefined refe                                                                                                                                                             rence to `ilclient_state_transition'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:103: undefined refe                                                                                                                                                             rence to `ilclient_state_transition'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:104: undefined refe                                                                                                                                                             rence to `ilclient_cleanup_components'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:108: undefined refe                                                                                                                                                             rence to `ilclient_destroy'
/usr/bin/ld: renderers/librenderers.a(audio_renderer_rpi.c.o): in function `audi                                                                                                                                                             o_renderer_rpi_init_renderer':
/home/pi/RPiPlay/renderers/audio_renderer_rpi.c:129: undefined reference to `ilc                                                                                                                                                             lient_create_component'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:145: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:145: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:178: undefined refe                                                                                                                                                             rence to `ilclient_setup_tunnel'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:191: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:191: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:212: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:212: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
/usr/bin/ld: /home/pi/RPiPlay/renderers/audio_renderer_rpi.c:227: undefined refe                                                                                                                                                             rence to `ilclient_get_handle'
[... cut ...] l'
/usr/bin/ld: /home/pi/RPiPlay/renderers/video_renderer_rpi.c:138: undefined refe                                                                                                                                                             rence to `ilclient_disable_tunnel'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rpiplay.dir/build.make:99: rpiplay] Error 1
make[1]: *** [CMakeFiles/Makefile2:206: CMakeFiles/rpiplay.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

Baza-86, 2022-09-07:
try with just make -j1, I kept getting the same error with the PLATFORM=rpi1 flag

visit1985, 2024-04-17:
#262 seems related.