FD-/RPiPlay #348: GEANT4 error

khadijach02, 2022-09-21:
Hello everyone, i really need your help 
I'm trying to create a project on the BNCT.
here's my output after make : 

Consolidate compiler generated dependencies of target test
[ 16%] Linking CXX executable test
/usr/bin/ld : CMakeFiles/test.dir/action.cc.o : dans la fonction « MyActionInitialization::Build() const » :
action.cc:(.text+0x5f) : référence indéfinie vers « MyPrimaryGenerator::MyPrimaryGenerator() »
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test.dir/build.make:208 : test] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:84 : CMakeFiles/test.dir/all] Erreur 2
make: *** [Makefile:91 : all] Erreur 2

any help??!!!!

pallas, 2022-09-21:
What are you talking about?

khadijach02, 2022-09-21:
@pallas i’m creating a project about BNCT 
Firstly, i creat a cell with an alpha inside…
I already did the construction folder, the action folder, the generator folder, the physics folder…
After running the program, the output of the command make is what i already shared

pallas, 2022-09-21:
It sounds like that doesn't have anything to do with this project.

closed 2022-09-21 by pallas