FD-/RPiPlay #349: ios 15.6.1 won't show up when connecting

AndyScripts, 2022-09-24:
I have two iphones, One on ios 12.4 and the other on ios 15.6.1. I'm able to mirror and see my iphone on 12.4. But when I use my 15.6.1 iphone , it connects but doesn't show anything mirroring, just black screen.

I use my iphone on ios 15.6.1 daily and was hoping maybe there is a work around since the legacy airplay only seems to work?

fduncanh, 2022-09-25:
works fine with iPhone SE 2nd gen. and iOS 16.0  (and worked with iOS 15.7)
"Legacy Pairing" used by rpiplay is still supported in iOS 16.

AndyScripts, 2022-09-25:
> works fine with iPhone SE 2nd gen. and iOS 16.0 (and worked with iOS 15.7)
> "Legacy Pairing" used by rpiplay is still supported in iOS 16.

Interesting input, I’ll update to iOS 16, I have a 2020 iPhone SE as-well. So we have the same phone I believe.

AndyScripts, 2022-09-25:
This can be closed, can confirm the update worked and iOS 16 iPhone SE 2020 works with rpiplay. Oddly iOS 15.6.1 does not.

fduncanh, 2022-09-26:
@AndyScripts 
you should close it yourself

closed 2022-10-01 by FD-