FD-/RPiPlay #350: some code files in lib don't have copyright headers

fduncanh, 2022-10-01:
maybe nitpicking, but it would be good to fix this

(dsafa22?)
lib/fairplay-playfair.c
lib/fairplay.h

raop.h
global.h

-------------------
(juvoh?)

dnssdint.h

-----------------
(FD-?)
dnssd.h

FD-, 2022-10-01:
It's been a while since I started the project, so I don't remember specifics. However, from some quick research, it seems like all of these files except dnssd.h stem from juvoh's shairplay, but don't bear a copyright notice there either.
dnssd.h was created by me a few years ago IIRC, but only as part of restructuring. I filled it with code from other parts (don't remember details; must've been either dsafa22's or juvoh's code).

fduncanh, 2022-10-01:
I'll do a PR here and fix in uxplay.

fduncanh, 2022-10-01:
The first part of dnssd.h is in shaitplay include , so that is also its origin

FD-, 2022-10-01:
I'm not sure about the legal perspective of someone other than the original author adding a copyright notice. If you want to clarify the source and license of the code, maybe just add something along the lines of "This code originates from the shairplay project by juhovh and contributors. It is used under permission of the ? license."

fduncanh, 2022-10-01:
Hmm.   OK.    Since you added it to RPiPlay,  you have some authorship, at least for changes.

```
/* This code originates from the shairplay project by juvoh and collaborators (2012-2013), 
/* https://github.com/juhovh/shairplay
/* used under permission of the GNU Lesser General Public License described below.
/* Modifications  Copyright (c) 2019  F. D. Raschbacher,  All Rights Reserved.
/ *
/* This library is free software; you can redistribute it and/or
/* modify it under the terms of the GNU Lesser General Public
/* License as published by the Free Software Foundation; either
/* version 2.1 of the License, or (at your option) any later version.
/*
/* This library is distributed in the hope that it will be useful,
/* but WITHOUT ANY WARRANTY; without even the implied warranty of
/* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
/* Lesser General Public License for more details.
 */
```

fduncanh, 2022-10-01:
submitted as PR #351

closed 2023-05-31 by fduncanh