FD-/RPiPlay #355: Raspberry PI Raspi 4 no Airplay output on HDMI ports

mschindl, 2023-01-28:
Hello,

the installation looks good so far, but I got no output from iOS 16.3 on the display/hdmi port.
HDMI cable is used in hdmi0 port.

I'm using following kernel 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l
```
pi@wohnzimmer:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
```

Installation steps:
```
sudo apt-get install cmake
sudo apt-get install libavahi-compat-libdnssd-dev
sudo apt-get install libplist-dev
sudo apt-get install libssl-dev

git clone https://github.com/FD-/RPiPlay.git
cd RPiPlay

# install the remaining plugins
$ sudo apt-get install -y libgstreamer1.0-dev \
     libgstreamer-plugins-base1.0-dev \
     libgstreamer-plugins-bad1.0-dev \
     gstreamer1.0-plugins-ugly \
     gstreamer1.0-tools \
     gstreamer1.0-gl \
     gstreamer1.0-gtk3

mkdir build
cd build
cmake ..
make -j
sudo make install
```

Logged in as user "pi":

```
$ rpiplay -n Wohnzimmer -a hdmi -ar gstreamer -l
Using low-latency mode
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 26
Local: 10.1.1.249
Remote: 10.1.1.161
Accepted IPv4 client on socket 33
Local: 10.1.1.249
Remote: 10.1.1.161
raop_rtp_mirror starting mirroring
```

Any ideas why the screen only shows the main desktop and changes not to the rpiplay input from iOS?

Other options were not helpful:

```
pi@wohnzimmer:~ $ rpiplay -b auto -a hdmi -l
Using low-latency mode
Initialized server socket(s)
Accepted IPv4 client on socket 31
Local: 10.1.1.249
Remote: 10.1.1.161
Accepted IPv4 client on socket 33
Local: 10.1.1.249
Remote: 10.1.1.161
raop_rtp_mirror starting mirroring
DRM_IOCTL_MODE_CREATE_DUMB failed: Cannot allocate memory
Failed to create scanout resource
DRM_IOCTL_MODE_CREATE_DUMB failed: Cannot allocate memory
Failed to create scanout resource
Segmentation fault

```

ABUCKY0, 2023-02-18:
Try specifying a different video renderer

mschindl, 2023-02-21:
Didn't work. Additional error will shown. It looks like that there is no easy way to get it on rpi4 working.

(Unfortunately I have now bought an Apple TV.)

closed 2023-02-21 by mschindl

ABUCKY0, 2023-02-21:
Mabye try using Raspberry Pi OS legacy (Not Bullseye). I got it working on my Raspberry Pi 4 Running Twister OS which I believe is based on the older kernel.

mschindl, 2023-02-21:
Good idea, but this Raspi provides a lot more services. Rpiplay as additional function would have been nice.

ABUCKY0, 2023-02-21:
I just read your error output, and it shows that it can't allocate enough memory, which could be your problem