FD-/RPiPlay #356: Nothing happens when trying to screenshare

Strimis10, 2023-02-18:
So when I start the program on my 3B (Raspian lite) this is what I get:

pi@raspberrypi:~/RPiPlay/build $ sudo ./rpiplay
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>

Then when I screenshare from my iphone (ios 15.7.2) I get this in the terminal:

pi@raspberrypi:~/RPiPlay/build $ sudo ./rpiplay
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 14
Local: 192.168.68.64
Remote: 192.168.68.59
Accepted IPv4 client on socket 16
Local: 192.168.68.64
Remote: 192.168.68.59
raop_rtp_mirror starting mirroring

which I assume is the right stuff, but there's no picture, like nothing at all, the iphone says that it's connected and obviously it is but nothing is shown on the monitor and there's no sound... I've read through a couple of threads that have similar issues but either they just say "oh it's fixed now" or there's no sulution

wvalcke, 2023-03-31:
I just see the exact same problem. Compiled version from master branch (11 Jan 2022)
Using raspberry PI4.

yrflx, 2023-04-03:
I had the same problem. did you find the solution or other option?

Fallcrim, 2023-04-03:
Same problem for me @FD- maybe you could check whether this is still supposed to work with newer iOS versions and stuff

JHthrowthegame, 2023-06-23:
Same problem here as well :(
I used this all the time a few years ago and wanted to start using it again.

samascience, 2023-07-14:
facing a similar issue !! 

installed this on ubuntu machine on linux. but the screen is showing nothing. 

Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>
Accepted IPv4 client on socket 31
Local: 10.0.0.246
Remote: 10.0.0.240
Accepted IPv4 client on socket 33
Local: 10.0.0.246
Remote: 10.0.0.240
raop_ntp receive timeout
raop_ntp receive timeout
raop_ntp receive timeout
raop_ntp receive timeout
Connection closed for socket 33
Destroying connection
Connection closed for socket 31
Destroying connection
Accepted IPv4 client on socket 31
Local: 10.0.0.246
Remote: 10.0.0.240
Accepted IPv4 client on socket 33
Local: 10.0.0.246
Remote: 10.0.0.240
^CStopping...
Removing connection for socket 31
Destroying connection
Removing connection for socket 33
Destroying connection

kohpai, 2023-07-14:
I think I have figured it out. The problem is with the port**s** used by RPiPlay. RPiPlay needs **TWO** ports to work. Below is how I used UxPlay on my Arch Linux:

```
uxplay -n "My Arch" -nh -p 30443,30444
```

If I used only one port, I can reproduce the problem, i.e. nothing happens after a connection. Therefore, I suggest anyone with this problem check your firewall first if it has the "allow some, deny all" policy. Basically, just make sure you have the two ports used by RPiPlay open. (I think by default these two ports are random)

DGulak, 2023-07-18:
I had the same problem. Looks like the're some renderer setup issue.
dummy audio and video renderers are default option
but when i try to start RPiPlay with parameters -vr rpi  i get "Unable to locate video renderer "rpi"."
when i try to start RPiPlay -v i get 
-vr renderer          Set video renderer to use. Available renderers:
    dummy: Dummy renderer; does not actually display video [Default]
-ar renderer          Set audio renderer to use. Available renderers:
    dummy: Dummy renderer; does not actually play audio [Default]
So the're nothing but dummy renderers for audion and video

crackley, 2023-10-30:
Try installing GStreamer, https://qengineering.eu/install-gstreamer-1.18-on-raspberry-pi-4.html, before building. Once I did this, gstreamer show up not only as an option but my default.

Strimis10, 2023-12-24:
I fixed it, installing GStreamer as the first thing worked, so on a clean os run:
`apt-get install libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libgstreamer-plugins-bad1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio`

Then:
`sudo apt-get install cmake
sudo apt-get install libavahi-compat-libdnssd-dev
sudo apt-get install libplist-dev
sudo apt-get install libssl-dev
sudo apt-get isntall git`

Then:
`git clone https://github.com/FD-/RPiPlay.git`

Then reboot.
and lastly:
`cd RPiPlay
mkdir build
cd build
cmake ..
make -j` 

aand:
`sudo make install`