FD-/RPiPlay #361: Screen is tearing with gstreamer

march1993, 2023-08-21:
Quite strage, problem happens only when vertical

[image]
Fig. Vertical Screen

[image]
Fig. Horizontal Screen

fduncanh, 2023-08-23:
maybe try [UxPlay](https://github.com/FDH2/UxPlay) (derived from RPiPlay)

dougg3, 2024-06-23:
I know I'm late to the party here but this looks very similar to [the problem I saw when running RPiPlay in a VMware or VirtualBox VM](https://www.downtowndougbrown.com/2021/04/finding-and-fixing-a-vmware-xorg-video-display-bug-using-gstreamer/). Are you running it inside of a VM? If so, this is actually a bug in the xf86-video-vmware plugin. A fix (version 13.4.0) was released early last year, but it's probably taking a while for various distros to pull it in. Looks like the fix was finally released with Ubuntu 24.04, for example.