FD-/RPiPlay #363: Display of Apple device not appearing

ruebenthepig73, 2023-09-24:
When I connect my iPhone or Mac to the RPiPlay server, it says that it has established the client and that it has started mirroring, but my phones display doesn't appear on the screen. I have tried to use VNC to see if I can get a display there, and I've even tried SSH to see if I can get a display that way. I am using a Raspberry Pi 4 and an iPhone running iOS 17.0.1

fduncanh, 2023-10-03:
You didn't give enough detail (e.g. OS,  RPiPlay video options etc.) for anyone to work out  your problem.

Try [UxPlay](http://github.com/FDH2/UxPlay), a RPiPlay descendant which is actively maintained.

ruebenthepig73, 2023-10-04:
Ok, I will try that. Thank you

GymLeaderTalon, 2024-01-11:
I had the same problem, I followed another users comment and it resolved the issue:

https://github.com/FD-/RPiPlay/issues/296#issuecomment-1678325787