FD-/RPiPlay #365: RPiPlay on Raspberry Pi 4 B 8B Issue

MikeTX01, 2023-11-03:
I'm new to all of this and have followed multiple installs on RPiPlay and cannot seem to get it to work.  I have followed the link but did not seem to find a fix.

I'm running the current OS from Raspberry Pi on my Raspberry Pi 4 B 8GB.

This is what i get:
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>

I did click the link.

What do i need to do to get it working?

I'm trying to mirror my iPad.

TIA

jmb-4, 2023-11-14:
I have the same issue, but couldnt fix it,

MikeTX01, 2023-11-16:
I finally got it to stream but it slowly lags more and more as the stream goes along.

KarlBaumann, 2023-11-17:
> I finally got it to stream but it slowly lags more and more as the stream goes along.

Why don't you tell us how did you fix it?

MikeTX01, 2023-11-19:
> > I finally got it to stream but it slowly lags more and more as the stream goes along.
> 
> Why don't you tell us how did you fix it?

I'm not running it in the latest version of Pi.   That was about all I did.  those warnings will pop up but it still mirrors.  it just becomes laggy as it continues though.

MikeTX01, 2023-11-20:
> Have you tried enabling low latency mode with -l on the command line? I don’t have a mirroring setup anymore but that’s what I used.

I have not and I'm not sure I can since I just click it to run.  

I'm new to all of this and I used PiKISS to install it. Is there some way to adjust it?