FD-/RPiPlay #369: Unable to stream on screen error ** (rpiplay:7237): CRITICAL **: 21:32:04.090: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

spicyynugget, 2023-12-26:
So, when i start the service i get GStreamer-CRITICAL errors.

 ./rpiplay -n rpiplay -a hdmi

(rpiplay:7237): GLib-GObject-WARNING **: 21:31:53.648: invalid cast from 'GstQueue' to 'GstBin'

(rpiplay:7237): GStreamer-CRITICAL **: 21:31:53.651: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(rpiplay:7237): GLib-GObject-WARNING **: 21:31:53.652: invalid cast from 'GstQueue' to 'GstBin'

(rpiplay:7237): GStreamer-CRITICAL **: 21:31:53.654: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(rpiplay:7237): GLib-GObject-WARNING **: 21:31:53.657: invalid cast from 'GstQueue' to 'GstBin'

(rpiplay:7237): GStreamer-CRITICAL **: 21:31:53.660: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(rpiplay:7237): GLib-GObject-WARNING **: 21:31:53.661: invalid cast from 'GstQueue' to 'GstBin'

(rpiplay:7237): GStreamer-CRITICAL **: 21:31:53.662: gst_bin_get_by_name: assertion 'GST_IS_BIN (bin)' failed

(rpiplay:7237): GLib-GObject-CRITICAL **: 21:31:53.664: g_object_set: assertion 'G_IS_OBJECT (object)' failed
Initialized server socket(s)
*** WARNING *** The program 'rpiplay' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/blog/projects/avahi-compat.html>

And as soon as i start to stream a video to the RPIplay the output goes nuts!

Accepted IPv4 client on socket 14
Local: 192.168.1.115
Remote: 192.168.1.55
Accepted IPv4 client on socket 16
Local: 192.168.1.115
Remote: 192.168.1.55
raop_rtp starting audio

** (rpiplay:7237): CRITICAL **: 21:32:03.673: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.679: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.697: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.700: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.768: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.772: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.775: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.778: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.782: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.791: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.793: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.795: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.820: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.824: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.826: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.861: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:03.864: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

[... 97 lines cut ...]
** (rpiplay:7237): CRITICAL **: 21:32:04.308: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:04.311: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:04.329: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:04.331: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed

** (rpiplay:7237): CRITICAL **: 21:32:04.333: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed
Removing connection for socket 14
Destroying connection

** (rpiplay:7237): CRITICAL **: 21:32:04.370: gst_app_src_push_internal: assertion 'GST_IS_APP_SRC (appsrc)' failed
Removing connection for socket 16
Destroying connection

** (rpiplay:7237): CRITICAL **: 21:32:04.395: gst_app_src_end_of_stream: assertion 'GST_IS_APP_SRC (appsrc)' failed

(rpiplay:7237): GStreamer-CRITICAL **: 21:32:04.399: gst_object_unref: assertion 'object != NULL' failed

(rpiplay:7237): GStreamer-CRITICAL **: 21:32:04.406: gst_object_unref: assertion 'object != NULL' failed

** (rpiplay:7237): CRITICAL **: 21:32:04.410: gst_app_src_end_of_stream: assertion 'GST_IS_APP_SRC (appsrc)' failed

Does anybody have any idea how to fix it?

fduncanh, 2023-12-28:
Try [UxPlay](http://github.com/FDH2/UxPlay) if you want to use GStreamer instead of OpenMax for video on R. Pi

GymLeaderTalon, 2024-01-11:
I followed this guys comment and it worked for me, the only thing I added was going into raspi-config and enabling auto login.

https://github.com/FD-/RPiPlay/issues/296#issuecomment-1678325787