FD-/RPiPlay #371: Not showing on Raspberry pi 32Bits

rijotech, 2024-01-21:
I successfully compiled it on a Raspberry Pi 5 32bits and executed but there is no display on the screen after connecting. I'm wondering why is not showing anything, also there are no errors.
DISPLAY:=0 rpiplay -n RASPLED -l -a hdmi -b off
```
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

rijotech, 2024-01-22:
I got it working on pi5 running on Debian 12 Bookworm in full-screen mode thanks to [this](https://github.com/FD-/RPiPlay/issues/218
) post. 
the only problem I'm facing is audio problems, there is no sound.
I'm running this command `rpiplay  -l -a hdmi -b off` I have also set the audio to `hdmi-0` on `raspi-config`
Any idea on how I can fix this?

fduncanh, 2024-01-29:
you might want to try  with RPiPlay-derived  [UxPlay](http://github.com/FDH2/UxPlay)