FD-/RPiPlay #372: Impossible to Screen mirror Netflix

Matheo5, 2024-02-05:
Hey ! 
Since 5 or 6 months, when I play a movie on Netflix website, it shows a black screen...
Does anyone know why ? Does Netflix now do not allow screen mirroring ? 
Thanks.

GoldenEye006, 2024-02-07:
Hey,
I guess this is something with DRM?
The video content is protected.

Matheo5, 2024-02-09:
So is their any solutions ?

Wayeet, 2024-06-03:
This might not be a problem to be handled by this software. The video content is protected and will also not work when streaming in a videoconference system or discord etc. This problem has been around for some time now and my personal solution is to use a browser that I can configure to bypass these restrictions. Streaming should also work again then. For some streaming services like for e.g. most of the tv media libraries it works by just using firefox without hardware acceleration.

Matheo5, 2024-07-28:
So which browser do you use ?

Matheo5, 2024-07-28:
Sorry I closed it i didn’t want it :)

Wayeet, 2024-07-28:
Should work just fine using the latest Firefox Browser 👍 @Matheo5

Matheo5, 2024-11-08:
Firefox can disable drm protections, but websites such as Apple TV, Netflix or tv channels can detect that your browser has disabled it... Still cannot screen mirror on my raspberry to watch series from subscribed platforms... Any idea ? Because I guess a real Apple TV box with support screen mirroring is able to bypass those drm...